winder/Universal-G-Code-Sender #2341: Linux

j00ch, 2023-10-16:
### Version

2.0.21

### Hardware / Firmware

GRBL 1.1

### What happened

Cannot build with openjdk17, openjdk13 succeeds.

```
[ERROR] Errors: 
[ERROR]   GUIBackendTest.setWorkPositionWithAdditionExpression:595 » NullPointer Cannot ...
[ERROR]   GUIBackendTest.setWorkPositionWithDivisionExpression:634 » NullPointer Cannot ...
[ERROR]   GUIBackendTest.setWorkPositionWithDivisionExpressionShouldConvertToWorkPositionUnits:673 » NullPointer
[ERROR]   GUIBackendTest.setWorkPositionWithDivisionExpressionWithoutValue:660 » NullPointer
[ERROR]   GUIBackendTest.setWorkPositionWithDivisionExpressionhouldConvertHashToWorkPositionUnits:647 » NullPointer
[ERROR]   GUIBackendTest.setWorkPositionWithExpressionShouldSetNegativePosition:582 » NullPointer
[ERROR]   GUIBackendTest.setWorkPositionWithExpressionShouldSetPosition:569 » NullPointer
[ERROR]   GUIBackendTest.setWorkPositionWithMultiplicationExpression:608 » NullPointer C...
[ERROR]   GUIBackendTest.setWorkPositionWithMultiplicationExpressionWithoutValue:621 » NullPointer
[ERROR]   GUIBackendTest.setWorkPositionWithSubtractionExpression:686 » NullPointer Cann...
[ERROR]   GUIBackendTest.setWorkPositionWithValueExpressionShouldSetPosition:556 » NullPointer
[INFO] 
```

### How to reproduce

Using Arch Linux.
jdk17-openjdk package installed.
Install package universal-gcode-sender from AUR.
Edit PKGBUILD
- update version number
- update java depends to java-17-openjdk
- update java home variable to /usr/lib/jvm/java-17-openjdk
run PKGBUILD

### Operating System

Arch Linux

breiler, 2023-10-16:
Are you using the latest master branch, or is it the v2.0.21 tag?

You must use the latest master branch as Java doesn't come with a JavaScript engine as of Java 16.

j00ch, 2023-10-16:
I'm currently using the nightly build from the releases page.

breiler, 2023-10-16:
Using Ubuntu, I installed `openjdk-17-jdk` and the `maven` package and ran `mvn clean install` and it compiled just fine.

Can you share the PKGBUILD script and the complete output from the build process?

j00ch, 2023-10-16:
Ok, using a zip of the master branch the build was successful. However that wouldn't work with the current PKGBUILD which is versioned: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=universal-gcode-sender

Will the next release include the resources currently missing to build with jdk17?

breiler, 2023-10-17:
Yes.

Up to version 2.0.21 you must use Java 13 to build.
Nightly and any following version you must use Java 17.

breiler, 2024-01-02:
Closing as we didn't get a response from the user.

closed 2024-01-02 by breiler