winder/Universal-G-Code-Sender #2342: IMPORTED SVGs BORDERS ON DESIGN PAGE

HomebuiltCNC, 2023-10-17:
### Version

2.0.21

### Hardware / Firmware

GRBL 1.1

### What happened

I imported an SVG file onto the design sheet and its aspect ratio was changed. I then set its width to the original 165mm width. The frame around the shape is 165mm but the shape itself is not 165mm wide. The The SVG has no border and no hidden/floating elements. I tried importing a different SVG (heart shape) and got the same result, however importing a circle or square had no problems. This is the screen shot showing the issue:

[image]

The SVG data is shown in this screen shot:
[image]

### How to reproduce

Import the attached SVG file onto the design sheet and the aspect ratio should be changed from the original 165 in the x direction and 145.115 in the y direction. Putting the left bottom corner onto (0,0) shows the white space on either side of the heart shape.

[image]

### Operating System

MAC OS Catalina Version 10.15.07

### Anything else

Every time the file is imported.

breiler, 2023-10-18:
Thanks for the report. I did a quick test and there is obviously something wrong when importing SVG:s. This will require some time to investigate...

HomebuiltCNC, 2023-10-27:
I have discovered that I can resolve my own issue by making sure that, in the SVG, I have a node at the left, right , top and bottom extremes of the path. These then ensure the SVG is brought into the editor with the correct sizes.
It appears UGS may look only at the coordinates provided within the SVG's bezier curve path and does not calculate the correct relevant left, right, top or bottom extremes?
I have now also found that UGS won't import SVG rectangles with rounded corners. 
I'm getting interested in possibly helping out here. Is there something I should/could do?

breiler, 2023-10-27:
I did some poking around and couldn't find anything obvious in the code. It may be the library we are using for importing or it may be something with my code converting to the internal model. It is tricky to debug and right now I have some more urgent issues to work with.

> I have now also found that UGS won't import SVG rectangles with rounded corners.
I'm getting interested in possibly helping out here. Is there something I should/could do?

You are very welcome to try, it would be a nice addition! 
To speed up development you can use the [DesignerMain](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-plugin-designer/src/main/java/com/willwinder/ugs/nbp/designer/DesignerMain.java) to start the designer tool. 

You might be able to add some properties for the rounded corners here to the rectangle shape:
[Rectangle.java](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-plugin-designer/src/main/java/com/willwinder/ugs/nbp/designer/entities/cuttable/Rectangle.java).