winder/Universal-G-Code-Sender #2356: Joystick

itsnotlevel1, 2023-10-26:
### Version

2.1.0

### Hardware / Firmware

FluidNC

### What happened

Downloaded pre-release....program opens and works so far but the joystick controls dont work. Meaning I can usually see the keys light up green so I know which button it corresponds but it doesnt light up green on any of the buttons.....connected to pc....same controller I was using before and it worked with the latest release 2.0.21. 

### How to reproduce

Open UGS and try to use joystick....from settings config....

### Operating System

MacOS 14 Sonoma

### Anything else

everytime.
 
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/13182782/messages.log)

itsnotlevel1, 2023-10-26:
Tried on desktop and laptop.....no worky.

breiler, 2023-10-29:
Thanks for the report, I have replaced the gamepad library which have the specific aarch64 library files. It is now available in the latest nightly build.

breiler, 2023-11-03:
Have you had the chance to test?

itsnotlevel1, 2023-11-04:
I did but im not sure I updated to the nightly as the "windows" machine I have my router table on is on this version of UGS:

Product Version: Universal Gcode Platform 20231026
Java: 17.0.8.1; OpenJDK 64-Bit Server VM 17.0.8.1+1
Runtime: OpenJDK Runtime Environment 17.0.8.1+1
System: Windows 11 version 10.0 running on amd64; Cp1252; en_US (ugsplatform)
User directory: C:\Users\Paul\AppData\Roaming\ugsplatform
Cache directory: C:\Users\Paul\AppData\Local\ugsplatform\Cache\var\cache

breiler, 2023-11-04:
> And it works there....I have been using the handheld controller and running gcode so it seems to be working fine.

Where is "there"? I was interested on having this tested on a Mac that is running on an ARM64 processor which I don't have access to. The fixes where applied on Oct 29, so just to be safe you will need to download a new fresh copy here: https://github.com/winder/Universal-G-Code-Sender#downloads

(If you have the time to just spin up a new version, activate the joystick and see that the buttons are highlighted in the joystick settings UI I would really appreciate it!)

> When I use the pendant the only thing that bugs me is changing the jog movement amount manually instead of maybe...

I created a new issue here for the pendant fixes that you described: #2360
Thanks for the feedback

> Does UGS keep the settings of how I configure the layout or joystick or anything I change myself to make it how I like it....I notice that if I uninstall and reinstall say for instance the nightly my layout and settings are persistent at least in MacOS, havent noticed it yet in Windows......where are the files for the persistent config settings? Hope Im saying that correct.

Yes, it will use the same settings directory for storing all configuration you make. The settings are stored here: 
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#configuration-files

itsnotlevel1, 2023-11-09:
I downloaded the latest for MacOS ARM 2.1.0 and no, the joystick does not work. :-(

I also downloaded the latest for Linux 2.1.0 (Ubuntu 22.04) and the joystick does not work there either. 

I am currently using UGS on a Windows 11 install and the Joystick DOES work with 2.1.0 Until, now UGS hangs in Windows after running a couple of commands or just leaving it open for a few minutes. I can look for other logs but Event viewer is the only place I know to look in Windows.....its only been doing this since today. 

- System 
  - Provider 
   [ Name]  Application Hang 
   [ Guid]  {c631c3dc-c676-59e4-2db3-5c0af00f9675} 
   EventID 1002 
   Version 0 
   Level 2 
   Task 101 
   Opcode 0 
   Keywords 0x8000000000000000 
  - TimeCreated 
   [ SystemTime]  2023-11-08T20:55:53.4655420Z 
   EventRecordID 3763 
   Correlation 
  - Execution 
   [ ProcessID]  9012 
   [ ThreadID]  7128 
   Channel Application 
   Computer RouterTable 
  - Security 
   [ UserID]  S-1-5-18 
- EventData 
  AppName ugsplatform64.exe 
  AppVersion 0.0.0.0 
  ProcessId 0xd18 
  StartTime 0x1da128082374738 
  TerminationTime 4294967295 
  ExeFileName C:\Users\Paul\Downloads\win64-ugs-platform-app-2.1.0\ugsplatform-win\bin\ugsplatform64.exe 
  ReportId d4147dc6-9cc0-46e8-a1fe-6cffe3608d95 
  PackageFullName  
  PackageRelativeAppId  
  HangType Top level window is idle

itsnotlevel1, 2023-11-09:
Ok, so this just happened in MacOS.....

[image]
[image]

I noticed a red dot in UGS for errors???

breiler, 2023-11-09:
The 2.1.0 will not work with gamepads on a Mac ARM 64-bit , I wanted you to try the nightly build.

Anyway, I have created a new build v2.1.1 that contains the gamepad fixes. I still don't know if it will work on MacOSX ARM64-bit though so I'd appreciate your feedback:
https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.1.1

Regarding the Windows crash, I don't have Windows 11 so I don't know if it is related to that. Let me know if the problem persists.

itsnotlevel1, 2023-11-09:
I grabbed the nightly (MacOS ARM) and everything looks great. The gamepad works as normally as one might expect. All buttons and pads and joysticks work. I ran a simple gcode file and it worked great.

With all the computers in my possesion..... Macbook air, Mac mini, 2 small NUC style PC's in which I run Ubuntu 22.04 on one and windows 11 on the other and One more windows 10 computer on my plasma table, I am able and trying to see which of UGS works best.
I know, it may seem like a lot but I was in IT for about 10 years, so I love it....at heart I am a Linux person.....Mint specifically but here recently have decided to try Ubuntu Cinnamon and so far, I like it.
With all that being said, I was wondering what you use 'OS' on the daily or with your CNC?
Another question would be, The GRBL + NETWORK usability of UGS.....viable or useful? And Im saying network as in local LAN not remote, that would be silly I think. Thoughts? What is the 'websocket' connection to UGS? Useful?
Thanks for being patient, I cant test exactly when I want and sometimes might not understand what you need from me from a testing perspective.

Paul

breiler, 2023-11-09:
> I grabbed the nightly (MacOS ARM) and everything looks great. The gamepad works as normally as one might expect. All buttons and pads and joysticks work. I ran a simple gcode file and it worked great.

Excellent, thanks for testing and for the feedback!

> With all that being said, I was wondering what you use 'OS' on the daily or with your CNC?

I have been using Linux up until around 2010 when I switched over to Mac. I have owned three different Macs and are now fed up with all of them as I am not able to upgrade the OS anymore. So I built an ordinary PC and switched back to Ubuntu Linux as my main OS (and dual boot to Windows 10). And my CNC-machine is using an old laptop running Ubuntu.

> Another question would be, The GRBL + NETWORK usability of UGS.....viable or useful? And Im saying network as in local LAN not remote, that would be silly I think.

I have personally not used the legacy GRBL over a LAN so I can't give any advice on that. But I have tested running both FluidNC, GrblHAL and Smoothieware over LAN and it seems to work ok. In theory it should provide a more stable connection as the TCP/IP protocol have some error handling. I know that there has been a lot of discussion about bandwith and latency causing bad performance for people running laser engravers on FluidNC. If that is caused by the specific type of gcode for laser engravers (which typically requires more bandwidth) or because of it running over WiFi I don't know.

> What is the 'websocket' connection to UGS? Useful?

You would need a controller that supports websocket. I have personally not tested this. But in my opinion you might as well use telnet or plain socket communication.

itsnotlevel1, 2023-11-09:
Thank you for all that!
I should have specified I am using FluidNC so I should have typed FluidNC + Network.

If I try to use TCP which would be badass as I wouldnt need the serial connection.....at least thats the idea.....I think. UGS can see the FluidNC 10.0.0.122 connection but when I try and connect the console gives me this:

*** Connecting to tcp://10.0.0.122:22
*** Fetching device status
*** Fetching device firmware version
*** Expected a 'FluidNC 3.3.0' or later but got 'Unknown 0.0.0'
*** Could not establish connection with the controller*** Connection closed

.....any ideas?

Paul

breiler, 2023-11-09:
Can you right click in the console an select "Enable verbose logging" and then connect.

breiler, 2023-11-09:
Nevermind I managed to reproduce it:
```
*** Connecting to tcp://192.168.1.10:23
*** Fetching device status
> ?
<Idle|MPos:0.000,0.000,0.000,0.000|FS:0,0|Pn:XYZ>
Other: ok
*** Fetching device firmware version
> $I
error:22
*** Expected a 'FluidNC 3.3.0' or later but got 'Unknown 0.0.0'
*** Could not establish connection with the controller
*** Connection closed
```

I can't reproduce it from a telnet client, so this requires some more digging. I'll let you know when it's fixed.

itsnotlevel1, 2023-11-09:
This is what I got from verbose logging.....

*** Connecting to tcp://10.0.0.122:22
*** Fetching device status
> ?
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
Other: ok
*** Fetching device firmware version
> $I
error:Gcode undefined feed rate
*** Expected a 'FluidNC 3.3.0' or later but got 'Unknown 0.0.0'
*** Could not establish connection with the controller
*** Connection closed

Paul

itsnotlevel1, 2023-11-19:
I realize you probably cant help with this but.....I was using ugs without fail last week. Everything worked perfectly and then today I open ugs and it just hangs. It did this before....a few weeks ago and now it does it with nightly build... 2.1.1 and 2.1.2 so it starts and just hangs not fully opening with the apple spinning ball of death.
My thoughts are that something is currupt but out of nowhere??? WTF. Java? I dont know, its tiring. I use this mac mini for honestly only fusion 360 now. I have 2 ipads and am on my 3rd macbook air....all with no problems EVER. I think this MacMini desktop unit is screwed in some way.
I have already cleared the hd and reinstalled MacOS and all updates but it wants to act up. I THINK I can say its probably NOT UGS but im not 100%. If you have any ideas that would be great but its ok if you dont, Im sure you have a lot on your plate.
Thanks for reading.
Maybe you can see something in the log that I dont understand?

Paul

> This is what I got from verbose logging.....
>
> *** Connecting to tcp://10.0.0.122:22
> *** Fetching device status
>> ?
> <Idle|MPos:0.000,0.000,0.000|FS:0,0>
> Other: ok
> *** Fetching device firmware version
>> $I
> error:Gcode undefined feed rate
> *** Expected a 'FluidNC 3.3.0' or later but got 'Unknown 0.0.0'
> *** Could not establish connection with the controller
> *** Connection closed
>
> Paul
>
>
>> Nevermind I managed to reproduce it:
>>
>> *** Connecting to tcp://192.168.1.10:23
>> *** Fetching device status
>>> ?
>> <Idle|MPos:0.000,0.000,0.000,0.000|FS:0,0|Pn:XYZ>
>> Other: ok
>> *** Fetching device firmware version
>>> $I
>> error:22
>> *** Expected a 'FluidNC 3.3.0' or later but got 'Unknown 0.0.0'
>> *** Could not establish connection with the controller
>> *** Connection closed
>>
>> I can't reproduce it from a telnet client, so this requires some more digging. I'll let you know when it's fixed.
-------------------------------------------------------------------------------
>Log Session: Sunday, November 19, 2023 at 2:45:45 PM Eastern Standard Time
>System Info: 
  Product Version         = Universal Gcode Sender 18-7d33005615b3c744564f3d38ab9ef14bf8c3ec44
  Operating System        = Mac OS X version 14.1.1 running on aarch64
  Java; VM; Vendor        = 17.0.8.1; OpenJDK 64-Bit Server VM 17.0.8.1+1; Eclipse Adoptium
  Runtime                 = OpenJDK Runtime Environment 17.0.8.1+1
  Java Home               = /Applications/Universal Gcode Sender.app/Contents/PlugIns/jdk-17.0.8.1+1-jre/Contents/Home
  System Locale; Encoding = en_US (ugsplatform); UTF-8
  Home Directory          = /Users/paul
  Current Directory       = /
  User Directory          = /Users/paul/Library/Application Support/ugsplatform
  Cache Directory         = /Users/paul/Library/Caches/ugsplatform/var/cache
  Installation            = /Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/java
                            /Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/bin
                            /Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/ugsplatform
                            /Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/etc
                            /Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/ide
                            /Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/extra
                            /Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/platform
                            /Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/platform
  Boot & Ext. Classpath   = 
  Application Classpath   = /Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/platform/lib/boot.jar:/Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/platform/lib/org-openide-modules.jar:/Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/platform/lib/org-openide-util-lookup.jar:/Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/platform/lib/org-openide-util-ui.jar:/Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/platform/lib/org-openide-util.jar
  Startup Classpath       = /Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/platform/core/asm-commons-9.5.jar:/Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/platform/core/asm-9.5.jar:/Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/platform/core/org-netbeans-libs-asm.jar:/Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/platform/core/org-openide-filesystems-compat8.jar:/Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/platform/core/org-openide-filesystems.jar:/Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/platform/core/asm-tree-9.5.jar:/Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/platform/core/core.jar:/Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/platform/core/core-base.jar:/Applications/Universal Gcode Sender.app/Contents/Resources/ugsplatform/ugsplatform/core/locale/core_ugsplatform.jar
-------------------------------------------------------------------------------
INFO [com.willwinder.ugs.nbp.core.lifecycle.ServiceStarter]: Starting UGS 2.0-SNAPSHOT / 2023-11-19
[... cut ...]
INFO [com.willwinder.universalgcodesender.utils.KeepAwakeUtils]: Moving the mouse location slightly to keep the computer awake.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated more than 10 times, further logs of this record are ignored until the log record changes.

breiler, 2023-11-20:
The log looks fine. This may very well be a Java + Mac thing. Unfortunately I have no way of testing this. It would be interesting if someone else is also experiencing this on a Mac with a Apple Silicone M1 processor.

breiler, 2024-03-15:
I am closing this as the original issue with the gamepad is resolved. Note that on Windows you need to upgrade to 2.1.6 as earlier versions have a memory leak that only effects that OS.

closed 2024-03-15 by breiler