winder/Universal-G-Code-Sender #2357: M30 Remover processor - not disabling

DushkaF, 2023-10-26:
### Version

2.1.0

### Hardware / Firmware

GRBL 1.1

### What happened

Hi! 
I make a GRBL fork and I want use M30 code. But UGS can't send it. I checked config files but I not find how fix it. Can you fix it?

### Operating System

Windows 11

breiler, 2023-10-27:
Open the firmware configuration file `c:/users/[your username]/.ugs/firmware_config/grbl.json` and change the value `optional` to `true` and `enabled` to `false`:
```
      {
        "name": "M30Processor",
        "enabled": false,
        "optional": true
      },
```

DushkaF, 2023-10-27:
Thank you! This was the only negative when choosing your software; other not dead programs do not have convenient functionality, unlike UGS

closed 2023-10-27 by DushkaF