winder/Universal-G-Code-Sender #2359: UGS Freezes in the middle of job

cah-98, 2023-10-30:
### Version

2.0.21

### Hardware / Firmware

GRBL 1.1

### What happened

Whenever i try to run a job it gets a few minutes in, different every time and then it just stops and keeps the spindle on. Still it reports the "Run" state while continuing to count the time in the Send Status counter. All controls are disabled and i have to restart the controller and reconnect to it to get it started again. For me this happens with all jobs i have tried so far. I have tried both in the nightly and 2.0.21 with GRBL 1.1f. The machine i am using is a Genmitsu 3020 pro max.

### How to reproduce

Start a job, so far for me any gcode seems to cause this issue, and then wait until eventually it freezes up.

### Operating System

Windows 11 Pro V22H2 64-bit

### Anything else

Here is the latest gcode i tried.
[x-axis-brackets_1-Pocket.zip](https://github.com/winder/Universal-G-Code-Sender/files/13205721/x-axis-brackets_1-Pocket.zip)

Here is my log file:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/13206325/messages.log)

breiler, 2023-10-30:
This is a common problem with cheap Arduinos with the CH340 USB serial drivers. But as you are running a more robust and better quality machine we can rule this out.

Are you using a proper shielded USB-cable that came with the machine? No extensions and no USB-hubs?

Are you running it with the spindle that came with the machine?

What happens if you try to run the machine without the spindle?

cah-98, 2023-10-30:
I am using an extension cable connected to the original usb cable. Could try it without the extension. The spindle is original and i will try without it running aswell.

breiler, 2023-10-30:
Personally I prefer USB. 

But you could very well use TCP/IP connection over WiFi if your controller has those capabilities (my Genmitsu Pro Max does not). I know that many people with capable controller do this, but I am a bit concerned that any RF interference from spindle and stepper would impact the WiFi signal quality, causing lower bandwidths. 🤷🏻‍♂️ 

Nice that you are up and running, I'm closing this issue.

closed 2023-10-30 by breiler