winder/Universal-G-Code-Sender #2360: Pendant fixes

breiler, 2023-11-04:
### Description

The pendant should have buttons for "larger" or "smaller" to change the step size button like on the main interface. Also then pendant still shows xy separate from z even though in the main interface is "use same jog for xy and z".

itsnotlevel1, 2023-11-09:
Also the Feed rate at the bottom wont let you put a number....the 1 is persistent.....you can put in a number but you have to backspace the 1 from the front of the number....
So if I change the feed speed.....touch the box, type in 20 it says 120 and its 120ipm so I have to backspace the 1 out. Hope that makes sense.

breiler, 2024-06-28:
I don't think that this is relevant anymore with the new web pendant. I feel that it is now very easy to change the feed rate settings with the new drop downs:
[image]

closed 2024-06-28 by breiler