winder/Universal-G-Code-Sender #2364: Error: This setting requires a computing speed of 2147483kHz which exceeds the limit of 30kHz

mhm9002, 2023-11-12:
### Version

2.0.21

### Hardware / Firmware

GRBL 1.1

### What happened

I have upload GRBL on my Ardunio, and for some reason, UGS insists on a high machine Z position (very high and beyond limits). The setup wizard doesn't save any updates I enter showing the following message, and any tests for the axis are disabled (most probably because one of them exceeds the limit).

The error is :
 This setting requires a computing speed of 2147483kHz which exceeds the limit of 30kHz

I tried to search the internet for an answer, but didn't find anything!

### How to reproduce

Upload (https://github.com/TGit-Tech/GRBL-28byj-48) on Arduino
Connect Ardunio to the port
Run UGS
Start setup wizard.

### Operating System

Windows, and Mac

breiler, 2023-11-12:
Sounds like you have upgraded GRBL which has a different eeprom memory layout. Try and reset your controller settings with the command $RST=$

mhm9002, 2023-11-12:
Thank you so much. This solved it.

closed 2023-11-12 by mhm9002