winder/Universal-G-Code-Sender #2370: Z Axis moves upward when joystick disconnects.

itsnotlevel1, 2023-11-20:
### Version

2.1.2

### Hardware / Firmware

FluidNC

### What happened

It happened twice this evening. I was sitting next to my machine and the joystick was active....I had been using it minutes before. The joystick times out after a bit....maybe 10 - 15 minutes? Info on xbox controller says its a 15 min timeout. And when that happens....the z axis moves upward and it appears to move the distance of the current jog setting is.... mine was at 1" both times and it sounded like 1". If I shut the joystick off manually, it doesnt do it. Log file attached. 

And I just let it time out again....15 min and my z axis moved up 1", but it doesnt do it when I manually turn it off. 

### How to reproduce

Just let the joystick timeout on its own and it will move the z axis. 

### Operating System

Linux Ubuntu 22.04 LTS

### Anything else

Twice today.  Again intentionally a third time a min ago.

breiler, 2023-11-20:
I was able to reproduce this problem. It seems that we receive some junk data when the controller goes offline. Working on a fix.

breiler, 2023-11-24:
I am closing this issue as I am no longer able to reproduce it. If the problem remains please reopen and provide log file.

closed 2023-11-24 by breiler