winder/Universal-G-Code-Sender #2377: Tool doesn't lift between pockets

Sofocletus, 2023-12-01:
### Version

2.1.2

### Hardware / Firmware

GRBL 1.1

### What happened

I'm having an issue where the tool path generated cuts through the material in between pockets, when the tool size is larger than a certain value. For example with a 1mm tool size there is a line connecting two pocket areas but with the toolpath set to 0.25 the tool lifts off the material correctly.
The design is from an imported SVG that looks OK in Inkscape, no open paths or broken nodes.
Any ideas on what may be going on?

[image]

[image]

breiler, 2023-12-05:
What software are you using as a CAM?

Sofocletus, 2023-12-06:
I generated the G-Code in UGS itself from an imported SVG design.
I replicated the issue in this file, it mostly happens when the tool diameter is relatively large.
[sample.zip](https://github.com/winder/Universal-G-Code-Sender/files/13575261/sample.zip)
[image]

closed 2023-12-06 by breiler

breiler, 2023-12-06:
I think that I got it working and is available in the latest nightly build.

PR #2378 (2023-12-06): Fixed problem with buffered shapes will generate connected single isolated lines
Fixed problem with buffered shapes will generate single isolated lines but there is no retract

From:
[image]

To:
[image]

Fixes: #2377