winder/Universal-G-Code-Sender #2382: USG doesn't recognize my .nc files

FleurdeLee, 2023-12-08:
### Version

Other

### Hardware / Firmware

GRBL 1.1

### What happened

I've been using the older version of UGS for several years with a Carve King CNC. Recently upgraded to a new Mac from a 2017 Mac. The new Mac has a M3 chip and Ventura 13.5 OS.  Finally figured out how to download JAVA to the new MAC and install a new version of UGS.  My version says 2.0-SNAPSHOT / 2022-11-04.  I can connect to my CNC and jog the machine with no issues.  When i go to import a file, the file window opens, but it doesn't show any of my .nc files to select from. Simultaneously, I can open Finder and see all the .nc files.  Some of these files were created with Fusion 360 and other with Carbide Create.  My CNC and the older version of UGS previously had zero issues recognizing these files and executing them.

Is there a reason this new version of UGS is not recognizing the .nc extension?  I do have (1) .txt file that it will recognize and import. This is the only .txt gcode file I have, as it was the test file sent when the CNC was purchased.

### Operating System

MacOS Ventura 13.5

breiler, 2023-12-08:
NC-files should be accepted.

Is there a reason you are using your own Java and a fairly old version of UGS?

You are using a snapshot build of UGS between 2.0.12 and 2.0.13. One of the changes we did back then was to upgrade a graphics library for OpenGL. Before this upgrade we needed a very specific version of Java on MacOSX (13.0.1_9) to make it work. If you are using Java version 16 you will however get other problems as they removed a couple of features. 

On top of that, the M3-processor is a whole new archithtecture which requires that you to download Java that is suited for that architecture. So choosing which Java to run is a delicate operation...

May I suggest that you download and try the latest version described here:
https://universalgcodesender.com/installing/

FleurdeLee, 2023-12-08:
Thank you for your quick reply! I installed Java 20 because I thought that was why the older version wouldn't run on the new computer. And the install of Java 20 did allow me to open the older version of UGS that I couldn't open before.  After reading your response, I downloaded UGS 2.1.2 and was able to connect but now it doesn't show any of my files.  I can't even navigate through the folders.

So, I decided to uninstall all versions of UGS and Java 20 to have a clean slate. With all versions removed and Java 20 uninstalled.  I reinstalled 2.1.2 and got the same result.  Everything else works... it pairs correctly, jogs, etc.  The folders option will open the window but I have folders to navigate through.

Any other suggestion?  Your assistance is greatly appreciated.  Thank you!

FleurdeLee, 2023-12-08:
**Correction** the folders option will open but there are no folders listed in the drop down menu to navigate through.

FleurdeLee, 2023-12-08:
Update... when opening the files option, and click on the drop down menu for file location and/or file type, the drop down menu for both is empty.  However, you can click in the empty window and the folders/file type options will populate and you can now navigate through the files.  Just a little 'buggy' but it at least I can see them.

However, still no change to the nc files.  It will not populate them with the rest of the files so they can be selected.

breiler, 2023-12-08:
Thanks, I think that I have narrowed the problem down to the "look and feel" theme. I think I have a fix on the way, hold on for ~20 minutes.

breiler, 2023-12-08:
Please try the latest nightly build here:
https://github.com/winder/Universal-G-Code-Sender#downloads

FleurdeLee, 2023-12-08:
Installed the new download and the folder option window is working as it should, except it still does not populate my nc files.

FleurdeLee, 2023-12-08:
Update... downloaded a test .nc file github and saved it to my download folder.  UGS showed the nc file in the download folder and I was able to open it with no issues.  So, I moved one of my nc files to the download folder and was able to access that one too.  Not sure why it wasn't recognizing the files in the other folders, but I copied them all to another folder and I can now access them all using UGS with no problem.

I will run one of the files later today to make sure all is good from the file to UGS to CNC.  Thank you so much for your help!

breiler, 2023-12-17:
This is a duplicate of this issue: #2383

And unfortunately there is nothing we can do about it in UGS. Seems to be an issue with Java and the MacOSX permission system. 

Closing this as a won't fix, sorry.

closed 2023-12-17 by breiler