winder/Universal-G-Code-Sender #2383: UGS 2.1.2 sees empty folders in MacOS Catalina

pmap1993, 2023-12-08:
### Version

2.1.2

### Hardware / Firmware

GRBL 1.1

### What happened

Good day, UGS works well when it's the older version, but I decided to use the newer version today.

I have tried giving Full Disk Access in the Security part of Settings and this is what I see. Screenshot below:
[image]

Please note that the version 2.0.20 is perfectly fine! Screenshot below:
[image]

This is a newly reformatted laptop, I'm using an old 2012 MacBook for CNC only.
Is there something wrong with my program? or I have done something wrong?

### How to reproduce

Use the Open File dialog using latest version 2.1.

### Operating System

macOS Catalina 10.15.7

pmap1993, 2023-12-08:
UGS 2.1 with Full Disk Access.

[image]

breiler, 2023-12-08:
Can you start UGS 2.1.2, click on the menu `Tool -> Open log directory` and then drag the `messages.log` file to github.

breiler, 2023-12-08:
Never mind, I managed to reproduce the error. I am looking into what happened.

pmap1993, 2023-12-09:
> Please try the latest nightly build here:
> https://github.com/winder/Universal-G-Code-Sender#downloads

Sorry for the delay, it gave a similar issue:
[image]

I wonder why.
I deleted all UGS versions for this Nightly Build test.
Also provided Full Disk Access for this build as well..

pmap1993, 2023-12-12:
If it makes a difference, when using UGS 2.0 for the first time (like a new download) in Mac, it asks 'it would like to access your Desktop folder'.. compared to the latest one where it does not.

breiler, 2023-12-12:
> I do apologize for the long post

No need to apologize, thanks for taking your time testing.

I booted up my old iMac and I was able to reproduce the problem. I have read about other Java based programs that experience the same problem. This seems to be a bug either in the access rights system in MacOSX or in the integration with Java.

I don't think that we can do much about this. You will simply have to place the files in a sub directory in your home folder. =(

pmap1993, 2023-12-12:
> No need to apologize, thanks for taking your time testing.
>
> I booted up my old iMac and I was able to reproduce the problem. I have read about other Java based programs that experience the same [...]

Thank you for trying to update it

breiler, 2023-12-17:
I am closing this as there is nothing we can do about it. 

Sorry.

closed 2023-12-17 by breiler