winder/Universal-G-Code-Sender #2385: Incorrect instruction received by Plotter

TravisBumgarner, 2023-12-10:
### Version

2.1.2

### Hardware / Firmware

GRBL 1.1

### What happened

I was trying to plot a gcode file. The gcode file was generated from a Python script that has generated many other gcode files I've used successfully. About 30,000 instructions in to plotting, the console showed the error in the screenshot below. It looks like the plotter received the instruction `M3S1000` but the instruction sent was `M3 1000`. I've included a screenshot and the logs.
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/13626547/messages.log)

[image]

### How to reproduce

I keep getting errors after UGS has already plotted thousands of instructions at random moments. 

### Operating System

macOS Sonoma 14.1.1

breiler, 2023-12-10:
It sounds like there is a communication problem where characters in the gcode stream is lost.

You can disregard from the missing space. This is just UGS trimming spaces as they are not required by the controller. But during transmission another character was likely lost causing the error.

What type of controller are you using? The cheap Chinese Arduinos usually use CH340 USB serial chip which are notorious for being unstable.

What USB cable are you using? Is it extended or plugged in to a USB hub?

TravisBumgarner, 2023-12-11:
Thanks for the reply!

I'm using the [Doesbot A4](https://www.amazon.com/Doesbot-Working-Handwriting-Structure-Assemble/dp/B0BHW97N9W/ref=sr_1_9) which probably falls under the "cheap Chinese Arduinos". Sounds like I'll need to look for a better controller. 

Might you have a recommendation for where I can shop for an upgrade? I'm not quite sure what to Google.

breiler, 2023-12-11:
There was not much information about the controller on that web page, and the images of the controller was too blurry to read the chip markings. I think that I can see that it begins with CH...
[image]

I have heard about people having success lowering the baud rate to 9600. But for this you need recompile GRBL. You should have received the source code for your controller with the machine. If not the company should supply you with it if you contact them.

You could also try another gcode sender that isn't so strict when it comes to communication errors.

TravisBumgarner, 2023-12-11:
Thanks for your help!

closed 2023-12-11 by TravisBumgarner