winder/Universal-G-Code-Sender #2388: Jogging continues indefinitely when ctrl key is lifted before arrow key

Matthew-G-S, 2023-12-14:
### Version

2.1.2

### Hardware / Firmware

GRBL 1.1

### What happened

UGS is set to only jog when the ctrl key is pressed with an arrow key, and it makes sense that the machine keeps jogging if the arrow key is kept depressed while the ctrl key is lifted. However, if I then lift the arrow key, it keeps jogging indefinitely until another jog command is given. Before I knew the reason, this caused a collision with the bed.

(I'm not actually sure what version of GRBL the machine is using. It was set up by a previous owner.)

### How to reproduce

Be ready to hit the emergency stop. Press the ctrl key and an arrow key. Keeping the arrow key depressed, release the ctrl key. Release the arrow key.

### Operating System

Windows 10 Pro 22H2

breiler, 2023-12-14:
Thanks for the report, I managed to reproduce the problem on Linux as well.

closed 2023-12-14 by breiler

breiler, 2023-12-14:
Please try the latest nightly build here: https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

PR #2390 (2023-12-14): Deactivate continuous jogging when a key is released
Should fix #2388