winder/Universal-G-Code-Sender #2393: Set doesn't set values in ui on any axis

sfmissionmark, 2023-12-15:
### Version

2.1.2

### Hardware / Firmware

GRBL 1.1

### What happened

I run UGS 2.1.3 or 2.1.2 on Ubuntu 22.04.3. If I use click the value of any axis it displays a dialog with a set button. I expected to be able to set this to a value. Clicking set does noting but close the window. I am in G54 if that matters.

### How to reproduce

Launch ugs from terminal as sudo. Connect to grbl mega 5x device. Unlock device state. Click Value on X axis. Enter new value. Click set. Value for x axis stays the same.

### Operating System

Ubuntu 22.04.3

### Anything else

I have disabled the B axis in the ui so I could see the A axis other then that I believe everything is default.

breiler, 2023-12-15:
Thanks for the report, something has gone wrong with the packaging where it can't find required libraries for the JavaScript engine. I am investigating...

breiler, 2023-12-15:
I looks like it is working now. 

It is available in the latest nightly build: https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

sfmissionmark, 2023-12-16:
Works great thank you.

closed 2023-12-16 by sfmissionmark