winder/Universal-G-Code-Sender #2396: UGS 2.1.3 Joystick not working

HenrykXXX, 2023-12-15:
### Version

2.1.2

### Hardware / Firmware

FluidNC

### What happened

I updated to 2.1.3 today, and my gamepad stopped working. It works fine with older version. Buttons are not working and analog joysticks don't work either. Is that known issue with new version?

### Operating System

Windows 10

breiler, 2023-12-15:
I did update the game controller database in the later release from here https://github.com/gabomdq/SDL_GameControllerDB.  It might have been removed from it for some reason.

If you run this tool and give me the mapping string I can add it separately:
https://www.generalarcade.com/gamepadtool/

HenrykXXX, 2023-12-19:
Sorry for the late response, here is mapping string for my pad:

03000000790000000600000000000000,G-Shark GS-GP702,a:b2,b:b1,back:b8,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,leftshoulder:b4,leftstick:b10,lefttrigger:b6,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b11,righttrigger:b7,rightx:a2,righty:a4,start:b9,x:b3,y:b0,platform:Windows,

breiler, 2023-12-19:
It is added and should be available in the latest nightly build:
https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

HenrykXXX, 2023-12-20:
It still doesn't work for me...
I fixed it by copying joystick plugin from my old version of UGS.

CzRobotics, 2024-03-12:
I have a similar problem but it concerns the control of JOG from the keyboard ... I will describe it in a separate thread

breiler, 2024-03-15:
It will now be possible to add custom gamepad mappings without the need to recompile. Available in the nightly builds and will be available in the next version 2.1.7