winder/Universal-G-Code-Sender #2399: [ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:3.0.0-M5:test (default-test) on project ugs-platform-app:

garyemiller, 2023-12-17:
### Version

2.1.2

### Hardware / Firmware

GRBL 1.1

### What happened

Trying to follow the install instructions:

dilbert /mnt/backup/archive/GRBL/Universal-G-Code-Sender # mvn install

````

[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  01:59 h
[INFO] Finished at: 2023-12-16T20:23:37-08:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:3.0.0-M5:test (default-test) on project ugs-platform-app: There are test failures.
[ERROR] 
[ERROR] Please refer to /mnt/backup/archive/GRBL/Universal-G-Code-Sender/ugs-platform/application/target/surefire-reports for the individual test results.
[ERROR] Please refer to dump files (if any exist) [date].dump, [date]-jvmRun[N].dump and [date].dumpstream.
[ERROR] The forked VM terminated without properly saying goodbye. VM crash or System.exit called?
[ERROR] Command was /bin/sh -c cd /mnt/backup/archive/GRBL/Universal-G-Code-Sender/ugs-platform/application && /opt/openjdk-bin-21.0.1_p12/bin/java -javaagent:/root/.m2/repository/org/jacoco/org.jacoco.agent/0.8.7/org.jacoco.agent-0.8.7-runtime.jar=destfile=/mnt/backup/archive/GRBL/Universal-G-Code-Sender/ugs-platform/application/target/jacoco.exec -jar /mnt/backup/archive/GRBL/Universal-G-Code-Sender/ugs-platform/application/target/surefire/surefirebooter17798012133394543113.jar /mnt/backup/archive/GRBL/Universal-G-Code-Sender/ugs-platform/application/target/surefire 2023-12-16T20-05-58_122-jvmRun1 surefire13100323951945716552tmp surefire_102331498012583113107tmp
[ERROR] Error occurred in starting fork, check output in log
[ERROR] Process Exit Code: 2
[ERROR] org.apache.maven.surefire.booter.SurefireBooterForkException: The forked VM terminated without properly saying goodbye. VM crash or System.exit called?
[ERROR] Command was /bin/sh -c cd /mnt/backup/archive/GRBL/Universal-G-Code-Sender/ugs-platform/application && /opt/openjdk-bin-21.0.1_p12/bin/java -javaagent:/root/.m2/repository/org/jacoco/org.jacoco.agent/0.8.7/org.jacoco.agent-0.8.7-runtime.jar=destfile=/mnt/backup/archive/GRBL/Universal-G-Code-Sender/ugs-platform/application/target/jacoco.exec -jar /mnt/backup/archive/GRBL/Universal-G-Code-Sender/ugs-platform/application/target/surefire/surefirebooter17798012133394543113.jar /mnt/backup/archive/GRBL/Universal-G-Code-Sender/ugs-platform/application/target/surefire 2023-12-16T20-05-58_122-jvmRun1 surefire13100323951945716552tmp surefire_102331498012583113107tmp
[ERROR] Error occurred in starting fork, check output in log
[ERROR] Process Exit Code: 2
[ERROR] 	at org.apache.maven.plugin.surefire.booterclient.ForkStarter.fork(ForkStarter.java:748)
[ERROR] 	at org.apache.maven.plugin.surefire.booterclient.ForkStarter.run(ForkStarter.java:305)
[ERROR] 	at org.apache.maven.plugin.surefire.booterclient.ForkStarter.run(ForkStarter.java:265)
[ERROR] 	at org.apache.maven.plugin.surefire.AbstractSurefireMojo.executeProvider(AbstractSurefireMojo.java:1314)
[ERROR] 	at org.apache.maven.plugin.surefire.AbstractSurefireMojo.executeAfterPreconditionsChecked(AbstractSurefireMojo.java:1159)
[ERROR] 	at org.apache.maven.plugin.surefire.AbstractSurefireMojo.execute(AbstractSurefireMojo.java:932)
[ERROR] 	at org.apache.maven.plugin.DefaultBuildPluginManager.executeMojo(DefaultBuildPluginManager.java:126)
[ERROR] 	at org.apache.maven.lifecycle.internal.MojoExecutor.doExecute2(MojoExecutor.java:328)
[ERROR] 	at org.apache.maven.lifecycle.internal.MojoExecutor.doExecute(MojoExecutor.java:316)
[ERROR] 	at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:212)
[ERROR] 	at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:174)
[ERROR] 	at org.apache.maven.lifecycle.internal.MojoExecutor.access$000(MojoExecutor.java:75)
[ERROR] 	at org.apache.maven.lifecycle.internal.MojoExecutor$1.run(MojoExecutor.java:162)
[ERROR] 	at org.apache.maven.plugin.DefaultMojosExecutionStrategy.execute(DefaultMojosExecutionStrategy.java:39)
[ERROR] 	at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:159)
[ERROR] 	at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject(LifecycleModuleBuilder.java:105)
[ERROR] 	at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject(LifecycleModuleBuilder.java:73)
[ERROR] 	at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build(SingleThreadedBuilder.java:53)
[ERROR] 	at org.apache.maven.lifecycle.internal.LifecycleStarter.execute(LifecycleStarter.java:118)
[ERROR] 	at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:261)
[ERROR] 	at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:173)
[ERROR] 	at org.apache.maven.DefaultMaven.execute(DefaultMaven.java:101)
[ERROR] 	at org.apache.maven.cli.MavenCli.execute(MavenCli.java:906)
[ERROR] 	at org.apache.maven.cli.MavenCli.doMain(MavenCli.java:283)
[ERROR] 	at org.apache.maven.cli.MavenCli.main(MavenCli.java:206)
[ERROR] 	at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(DirectMethodHandleAccessor.java:103)
[ERROR] 	at java.base/java.lang.reflect.Method.invoke(Method.java:580)
[ERROR] 	at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced(Launcher.java:283)
[ERROR] 	at org.codehaus.plexus.classworlds.launcher.Launcher.launch(Launcher.java:226)
[ERROR] 	at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode(Launcher.java:407)
[ERROR] 	at org.codehaus.plexus.classworlds.launcher.Launcher.main(Launcher.java:348)
[ERROR] 
[ERROR] -> [Help 1]
````

### How to reproduce

mvn install

### Operating System

Gentoo amd64 unstable

garyemiller, 2023-12-17:
I tried this from git head.  Not the release.

garyemiller, 2023-12-17:
Build log.

garyemiller, 2023-12-17:
Build log

garyemiller, 2023-12-17:
Odd, I can't seem to upload the build log.

garyemiller, 2023-12-18:
[build.log.gz](https://github.com/winder/Universal-G-Code-Sender/files/13698245/build.log.gz)

This worked.

breiler, 2023-12-18:
The build log was not complete, so I don't know exactly what is going on. It is complaining about `Unsupported class file major version 65` which indicates to me that you are trying to compile using Java 21. We currently require Java 17.

Which instructions did you follow for compiling?
https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#development

Do you really need to compile the software? We have a prepackaged variant that comes with a Java runtime. No need to install it separatly, instructions can be found here:
https://universalgcodesender.com/installing/

garyemiller, 2023-12-18:
> The build log was not complete

I did this: nohup mvn 

And waited for it to finish.  Absolutely nothing else we output on stdout or stdin.

> Indicates to me that you are trying to compile using Java 21. We currently require Java 17.

Yes, I have openjdk 21, not openjdk 17.  I'l try again with 17.

> Which instructions did you follow for compiling?

From the REAME.md:

````
#### Compiling and starting the application
[...]

UGS Platform: 
```
bash
mvn install
mvn nbm:run-platform -pl ugs-platform/application
````

> Do you really need to compile the software?

Dunno.  I figured the git head was the latest and best.

> No need to install it separatly, instructions can be found here:

I looked at that.  Looks nothing like my Linux, so not useful to me.

I run Gentoo with Xfce desktop.

breiler, 2023-12-18:
The compile instructions is more for development, the build takes a long time and you will not get a package that you can run from.

> Dunno. I figured the git head was the latest and best.

It is also built in the nightly/snapshot builds here: https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

> I looked at that. Looks nothing like my Linux, so not useful to me.
I run Gentoo with Xfce desktop.

You should only need to run the `bin/ugsplatform` script if you download a prepackaged version.

garyemiller, 2023-12-18:
> takes a long time and you will not get a package that you can run from.

It worked with openjdk 17, in 13 minutes and 22 seconds.  Nothing compared to the hours that Chrome takes.

But, your are right, it does not run, I suspect not the way you expected:

````

dilbert /mnt/backup/archive/GRBL/Universal-G-Code-Sender # mvn nbm:run-platform -pl ugs-platform/application
[INFO] Scanning for projects...
[INFO] 
[INFO] ------------------< com.willwinder:ugs-platform-app >-------------------
[INFO] Building ugs-platform-app 2.0-SNAPSHOT
[INFO]   from pom.xml
[INFO] --------------------------[ nbm-application ]---------------------------
[INFO] 
[INFO] --- nbm:4.8:run-platform (default-cli) @ ugs-platform-app ---
[INFO] Executing: /bin/sh -c '/mnt/backup/archive/GRBL/Universal-G-Code-Sender/ugs-platform/application/target/ugsplatform/bin/ugsplatform' '--userdir' '/mnt/backup/archive/GRBL/Universal-G-Code-Sender/ugs-platform/application/target/userdir' '-J-Dnetbeans.logger.console=true' '-J-ea' '--branding' 'ugsplatform'
[INFO] Cannot find java. Please use the --jdkhome switch.
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  2.168 s
[INFO] Finished at: 2023-12-18T13:39:50-08:00
[INFO] ------------------------------------------------------------------------
dilbert /mnt/backup/archive/GRBL/Universal-G-Code-Sender # 
````

What do I need in addition to openjdk??

garyemiller, 2023-12-18:
I tried the nightly.  Started right up.  Nice.

It did throw these warnings:

````
dilbert /var/tmp/ugsplatform-linux-x64 # bin/ugsplatform 
OpenJDK 64-Bit Server VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will likely be removed in a future release.
WARNING: A terminally deprecated method in java.lang.System has been called
WARNING: System::setSecurityManager has been called by org.netbeans.TopSecurityManager (file:/var/tmp/ugsplatform-linux-x64/platform/lib/boot.jar)
WARNING: Please consider reporting this to the maintainers of org.netbeans.TopSecurityManager
WARNING: System::setSecurityManager will be removed in a future release
````

So you can add gentoo Unstable to the list of distros that work fine with the nightlies.

breiler, 2023-12-19:
> What do I need in addition to openjdk??

When rolling UGS on your own you need to change a config that points out the Java runtime location:
https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/application/pom.xml#L734C106-L734C163

And you would probably want to make a package instead of building with maven each time:
https://github.com/winder/Universal-G-Code-Sender/blob/master/.github/workflows/release.yaml#L41

It is not as easy as "./configure && make && make install"... =(

Nice that you are up and running, I am closing this.

closed 2023-12-19 by breiler

garyemiller, 2023-12-19:
Thanks for your help!

garyemiller, 2023-12-19:
[image]

This is my screen.  Notice how much smaller the Welcome Page text is.  x-height is only 6 pixels.

If it was the same, or a little smaller, than the menu items it would be fine.  As it is, even with pixeltool, it is hard for me to read.

garyemiller, 2023-12-19:
Oh, sorry, wrong issue.