winder/Universal-G-Code-Sender #2400: loading issue

kiddo2003, 2023-12-18:
### Version

Other

### Hardware / Firmware

GRBL 1.1

### What happened

if have reinstalled uge platform 2.1.3 over 20 times ,i have followed the installation instructions although its not working. the software starts loading for 2 to 3s after its dies out. is there is any solutions.

https://github.com/winder/Universal-G-Code-Sender/assets/154241717/d3c985db-0d08-4642-8394-a28cacf8d609

log of ugs platform uploaded below:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/13705390/messages.log)

[image]

### How to reproduce

follows the steps as follows in the below video link
https://youtu.be/3OdHT98wiNs

### Operating System

 Windows 11 Version 23H2 for x64-based Systems (KB4023057)

### Anything else

i have just now started to learn about cnc. its the first time im installing  it.

breiler, 2023-12-18:
Thanks for the detailed report!
I can't find any errors in the log, but it is missing alot of expected messages.

You could try and delete the config directory, it maybe got corrupt:
`C:\Users\ANUMESH\AppData\Roaming\ugsplatform`

If that doesn't work, I know that the platform we were using was sensitive to special characters in the directory path. 

Rename the folders with parantheses and remove those characters, from:
`C:\Users\ANUMESH\Desktop\New folder (2)\win64-ugs-platform-app-2.1.3 (2)\`

To:
`C:\Users\ANUMESH\Desktop\New folder 2\win64-ugs-platform-app-2.1.3\`

kiddo2003, 2023-12-20:
> Thanks for the detailed report! I can't find any errors in the log, but it is missing alot of expected messages.
>
> You could try and delete the config directory, it maybe got corrupt: [...]

hey, thank you for the help its working now(deleted the config directory followed the above steps )

closed 2023-12-20 by kiddo2003