winder/Universal-G-Code-Sender #2401: The "Welcome Page" has 6 pixel high text.

garyemiller, 2023-12-18:
### Description

The "Welcome Page" has 6 pixel high text.

I have vision issues, and 6 pixel high text is very hard for me to read.  Any way to fix that?

Other object, like the Menu test are fine.

breiler, 2023-12-19:
What text are you referring to? The text is (unfortunatley) hard coded, but the bread text is set to 11px:

[image]

garyemiller, 2023-12-19:
[image]

This is what I see.  The x-height in the Welcome Page is only 6 pixels.  Hard for me to read even with pixeltool.

If the text was the same size, or a tad smaller than the menu font, it would be fine.

garyemiller, 2023-12-22:
I found the problem.  You css is specifying fonts in pixels (px) and not points (pt).

For example:

./ugs-pendant/src/main/webapp/node_modules/unique-filename/coverage/base.css
  
  font-size: 14px;

At 96 DPI: 1 Point = 0.75 * Pixel

So 14px is 10.5 point type.  Not bad.

But, at 192 DIP: 1 Point = 0.375 * Pixel

So 14 px is 5.2 points!  That is pretty small!

If you specify everything in points (pt), instead of pixels (px), the everyone see the exact same size of font, regardless of the DPI of their screen.