winder/Universal-G-Code-Sender #2402: Error while processing response <ok>: An unexpected command was completed by the controller. ignore

francescoforcellini, 2023-12-23:
### Version

2.1.4

### Hardware / Firmware

GRBL 1.1

### What happened

I'm using a pen plotter equipped with arduino and a cnc shield, it's a corexy, now and then i receive the error posted as title.
This stops with the pen down and ruins the drawing... anyone can help me? I will be happy also to have just the machine stop with the pen up, instead of down... I have te log file as suggested in troubleshooting, I wil post the last few lines if is enough...
Thanks in advance!

INFO [null]: Last record repeated again.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Sending gcode file (C:\Users\franc\AppData\Local\Temp\1703352961746-0\aranciooutput.gcode_ugs_1703352961747).
INFO [UGSEventDispatcher]: Removing UGSEvent listener: SentCommandsHighlightContainer
INFO [com.willwinder.universalgcodesender.utils.KeepAwakeUtils]: Moving the mouse location slightly to keep the computer awake.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated more than 10 times, further logs of this record are ignored until the log record changes.
INFO [null]: Last record repeated 26 times in total.
SEVERE [com.willwinder.universalgcodesender.GrblController]: Error while processing response <<Run|WPos:190.390,119.550,0.000|Bf:15,14|FS:0,:192.300,116.020,0.000|Bf:15,9|FS:0,0|Pn:XYZ>>: For input string: ":192.300"
java.lang.NumberFormatException: For input string: ":192.300"
	at java.base/jdk.internal.math.FloatingDecimal.readJavaFormatString(Unknown Source)
	at java.base/jdk.internal.math.FloatingDecimal.parseDouble(Unknown Source)
	at java.base/java.lang.Double.parseDouble(Unknown Source)
	at com.willwinder.universalgcodesender.GrblUtils.getStatusFromStatusStringV1(GrblUtils.java:421)
	at com.willwinder.universalgcodesender.GrblUtils.getStatusFromStatusString(GrblUtils.java:336)
	at com.willwinder.universalgcodesender.GrblController.handleStatusString(GrblController.java:568)
[catch] at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:234)
	at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:772)
	at com.willwinder.universalgcodesender.communicator.event.CommunicatorEventDispatcher.lambda$rawResponseListener$0(CommunicatorEventDispatcher.java:60)
	at java.base/java.lang.Iterable.forEach(Unknown Source)
	at com.willwinder.universalgcodesender.communicator.event.CommunicatorEventDispatcher.rawResponseListener(CommunicatorEventDispatcher.java:60)
	at com.willwinder.universalgcodesender.communicator.event.AsyncCommunicatorEventDispatcher.dispatchEvent(AsyncCommunicatorEventDispatcher.java:126)
	at com.willwinder.universalgcodesender.communicator.event.AsyncCommunicatorEventDispatcher.run(AsyncCommunicatorEventDispatcher.java:78)
	at java.base/java.lang.Thread.run(Unknown Source)
WARNING [org.openide.util.ImageUtilities]: Initial slashes in Utilities.loadImage deprecated (cf. #20072): /org/netbeans/core/windows/resources/error.png
SEVERE [com.willwinder.universalgcodesender.GrblController]: Error while processing response <ok>: An unexpected command was completed by the controller.
com.willwinder.universalgcodesender.AbstractController$UnexpectedCommand: An unexpected command was completed by the controller.
	at com.willwinder.universalgcodesender.AbstractController.commandComplete(AbstractController.java:754)
[catch] at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:181)
	at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:772)
	at com.willwinder.universalgcodesender.communicator.event.CommunicatorEventDispatcher.lambda$rawResponseListener$0(CommunicatorEventDispatcher.java:60)
	at java.base/java.lang.Iterable.forEach(Unknown Source)
	at com.willwinder.universalgcodesender.communicator.event.CommunicatorEventDispatcher.rawResponseListener(CommunicatorEventDispatcher.java:60)
	at com.willwinder.universalgcodesender.communicator.event.AsyncCommunicatorEventDispatcher.dispatchEvent(AsyncCommunicatorEventDispatcher.java:126)
	at com.willwinder.universalgcodesender.communicator.event.AsyncCommunicatorEventDispatcher.run(AsyncCommunicatorEventDispatcher.java:78)
	at java.base/java.lang.Thread.run(Unknown Source)
SEVERE [null]: Last record repeated again.
INFO [com.willwinder.universalgcodesender.utils.KeepAwakeUtils]: Moving the mouse location slightly to keep the computer awake.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [null]: Last record repeated again.
INFO [org.netbeans.core.ui.focus]: External Changes Refresh on focus gain disabled

### Operating System

Windows 10

breiler, 2023-12-24:
The log tells me that your USB connection is unstable. UGS complains that it can't parse the status message: `<Run|WPos:190.390,119.550,0.000|Bf:15,14|FS:0,:192.300,116.020,0.000|Bf:15,9|FS:0,0|Pn:XYZ>`

Should probably have been the following. The response from your machine is missing at least 19 characters from the response.
```
<Run|WPos:190.390,119.550,0.000|Bf:15,14|FS:0,0|Pn:XYZ>
<Run|WPos:192.300,116.020,0.000|Bf:15,9|FS:0,0|Pn:XYZ>
```

Make sure that you are using a proper USB cable, no extensions and no USB-hub. 
If you are using a cheaper Arduino clone you could try and change the configuration in the `config.h` and lower the baud rate to `9600` to see if that helps: https://github.com/gnea/grbl/blob/master/grbl/config.h#L42

francescoforcellini, 2023-12-24:
Thank you very much Breiler for the quicl reply.
I tried with different cables, now I've changed the usb port and seems that is working well! I'm using and arduino clone, so I'm gonna change the baud rate as well.
Thank you!

closed 2023-12-24 by francescoforcellini