winder/Universal-G-Code-Sender #2406: Mac file browser issue

ImolaS3, 2023-12-29:
### Version

2.1.4

### Hardware / Firmware

GRBL 1.1

### What happened

I have been using UGS for a while on 2 different Macs and it won't display any files in any folder except for the default root folder.
The file list window just appears blank when i select any other folder

In addition sometimes, when i attempt to change directories, the "options" for directories n the top textbox are all blank

I have the latest version installed (just downloaded)

### How to reproduce

On a Mac running OS Monteray, just try to open  file from various directories 

### Operating System

Mac OS Monterary

### Anything else

Every time I use UGS (daily at present) and it has been happening for the full 56 weeks I have been using UGS

breiler, 2023-12-30:
Same issue as #2383. 

I doubt that we can solve this fully, I did however read somewhere that the problem is common with the `JFileChooser` and that the `FileChooser` might work better. I'll give that a try once I get some time over.

JayZmake, 2024-01-27:
I am completely new to this forum. This product looks great, but, the issue I have looks most like this one posted. I have successfully opened gcodesender on my Mac running Monterey12.6.2. However, I am not able to open any Gcode files I have created and view them. The file directories all show up as empty for all files on my Mac. Please excuse my newbeeness here, and let me know if and how I can forward what you might need to resolve this issue to get me up and running. Here is a screen shot of what happens:
[image]

breiler, 2024-01-29:
I have tried replacing the `javax.swing.JFileChooser` with an older `java.awt.FileDialog` which seems to work better on MacOSX. There are still some small issues on Linux which still needs to be fixed. There are many places where the `JFileChooser` is still used which needs to be replaced as well.

If someone is interested in testing it is available in the latest nightly build:
https://github.com/winder/Universal-G-Code-Sender/releases/tag/nightly

JayZmake, 2024-01-30:
The nightly build edit solved the problem! Thanks so much. Thank goodness there’s smart and hardworking people like yourself working on this stuff. I don’t know how you do it, but, it frees up my time to work on what I want to work on and I appreciate it.

closed 2024-02-08 by breiler