winder/Universal-G-Code-Sender #2411: Empty g-code output

garyemiller, 2024-01-01:
### Version

2.1.4

### Hardware / Firmware

GRBL 1.1

### What happened

On both 2.1.3 and the current nightly.

I load this simple ugsd file:

[rectangle.ugsd.txt](https://github.com/winder/Universal-G-Code-Sender/files/13802633/rectangle.ugsd.txt)

I see my rectangle, but the visualizer goes blank. .  I Export Gcode, and I get this empty g-code file:

[rectangle.gcode.txt](https://github.com/winder/Universal-G-Code-Sender/files/13802634/rectangle.gcode.txt)

What is wrong?

### Operating System

Gentoo

### Anything else

Every time

breiler, 2024-01-01:
I think that you have found a bug.

In the tool settings Window, make sure to set the depth per pass larger than zero:
[image]

After that you can click on the shape and set a target depth and then save:
[image]

garyemiller, 2024-01-01:
That worked.  Thanks for the quick answer.  Since this is a laser, and I have it focused, I really don't want Z changing.  This has come up in the "Discussions" area as well.

If you get a fix in nightly, I'll be happy to test it.

garyemiller, 2024-01-01:
That solution works, but doubles the number of cuts, first at Z0 and then again at Z0.1:

````
G1 X0.527Y-0.527Z-0
G1 X0Y-0.745Z-0
G0 Z5
G0 X0Y-0.133
G0 Z0
G1 F400 X0Y-0.133Z-0.1
G1 F1000G1 X0.527Y-0.527Z-0
G1 X0Y-0.745Z-0
G0 Z5
G0 X0Y-0.133
G0 Z0
G1 F400 X0Y-0.133Z-0.1
G1 F1000 X0Y-0.133Z-0.1
 X0Y-0.133Z-0.1
````

The lift for the move "G0 Z5", then back to "G1 Z0" is slightly overdone, but not bad.

And another problem now.

Here is the start of the output g-code:

````
; This file was generated with "Universal Gcode Sender 2.0-SNAPSHOT / 2023-12-19
"
;
; Tool: 0.5mm
; Depth per pass: 1.0mm
; Feed speed: 1000mm/min
; Plunge speed: 400mm/min
; Safe height: 5.0mm
; Tool step over: 0.3mm
; Spindle speed: 254rpm

G21 ; millimeters
G90 ; absolute coordinate
G17 ; XY plane
G94 ; units per minute feed rate mode
M3 S254 ; Turning on spindle

; Text - Pocket (1/2)
G0 Z5
G0 X44.433Y13.22
G0 Z0
G1 F400 X44.433Y13.22Z-0
G1 F1000 X44.433Y13.22Z-0
````

I have a laser on a new CNC 3018.  I'm connecting as "GRBL ESP32".  Not sure the difference bewteen "GRBL" and "GRBL ESP32".

That "M3 S254" turns on the laser, and leaves it on.  It should be "M4 S254",  M4 turns off the laser for G0, and back on for G1.  That I can fix by hand in the g-code file.

garyemiller, 2024-01-01:
Also, this is funcky, "Z-0":

````
G1 X0Y-0.745Z-0
````

breiler, 2024-01-02:
> I have a laser on a new CNC 3018. I'm connecting as "GRBL ESP32". Not sure the difference bewteen "GRBL" and "GRBL ESP32".

I would recommend you to run with the plain GRBL profile. GRBL ESP32 was for the now deprecated project https://github.com/bdring/Grbl_Esp32 (the new project is named FluidNC). The main difference is the startup sequence. I will probably remove the GRBL ESP32 profile in the future.

> Also, this is funcky, "Z-0"

Yes this is a small oddity with how float numbers are handled. It should probably be cleaned up in the future. This has no effect on the machine movement.

I have fixed it in the nightly build so that it now only generates one cut if you leave the start depth and target depth equal to each other.

closed 2024-01-14 by breiler