winder/Universal-G-Code-Sender #2412: File Save not always present

garyemiller, 2024-01-02:
### Version

2.1.4

### Hardware / Firmware

GRBL 1.1

### What happened

I am tweaking a ugsd file with text.  When I change something in a text block, like the X position, the "File Save" options stays grayed out.  I try to "File Save As" over the file, and that refuses to over write.

My work around is to "File Save As" to a tmp file.  Then "File Save As" over the original file.  That raises a warning box, but I can click that away.

Any way to ensure "File Save" is always active?  Or to make it active?

This is on the nightly.

### Operating System

Gentoo

breiler, 2024-01-02:
I think I fixed most of the places where this happened. It is still not perfect but is at least more usable.

garyemiller, 2024-01-02:
Is it in the current nightly?

breiler, 2024-01-02:
Yes it is available in the latest nightly build:
https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

garyemiller, 2024-01-02:
I just loaded the latest nightly.  If I select a text block, and change just the font name, I still can not select "File Save".  Changing just the "Target Depth" is now OK.

breiler, 2024-01-02:
Yes, the font name was a tricky to fix. That needs to wait until tomorrow.

garyemiller, 2024-01-02:
I can easily wait that long.  Thanks!

breiler, 2024-01-22:
This was a lot of work but I think that it is working ok now. 
I had to rewrite the whole settings functionality for the designer.

It is available in the latest nightly build.
https://github.com/winder/Universal-G-Code-Sender/pull/2434

I am closing this, but feel free to add a comment if it is still not working properly.

closed 2024-01-22 by breiler

garyemiller, 2024-01-22:
Thanks!  I'll try it now.  All that work to gray out the button.  Is there any good reason to not let the user save at any time?

breiler, 2024-01-22:
Nah, the button could have been enabled all the time it would not make much of a difference.

If it should be enabled or disabled is based on if the Undo-stack has changed, so if there is an edit which is not added to the stack things will get weird when using undo/redo. Probably destroying the design.

We also use this feature to warn the user if the file has not been saved to prevent them from sending the old file by mistake.

So the greyed out save button was the symptom not the cause and I would have had to deal with this sooner or later..