winder/Universal-G-Code-Sender #2413: UGS Screen Freeze and/or close out

Username-CNC, 2024-01-02:
### Version

2.0.11

### Hardware / Firmware

GRBL 1.1

### What happened

Execution
2.0.11
The problem also exists with version 2.1.2

Hardware/Firmware
GRBL 1.1

What happened:
When running the 64-bit module, the program exits on its own without any warning or error code. The spindle continues to run endlessly. I went into ugsplatform.conf and changed the value of Xms from 64 to 512. This had no effect on the problem I'm having. I tried running the 32 bit version and it will complete the project. However, the program would freeze during the project and all buttons would become unresponsive. I would have to quit the program.

How to reproduce
This problem can be easily reproduced on my computer.

operating system
Notebook Fujitsu
Windows 10 Home version 22H2
64-bit operating system
8GB RAM
Intel graphics

Something else
I use Vectric Vcarve Pro.

### Anything else

The problem occurs every time

breiler, 2024-01-02:
Can you post the log file:
C:/Users/[your username>]/AppData/Roaming/ugsplatform/var/log/messages.log

Username-CNC, 2024-01-02:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/13811092/messages.log)
[messages_1.log](https://github.com/winder/Universal-G-Code-Sender/files/13811093/messages_1.log)
[messages_2.txt](https://github.com/winder/Universal-G-Code-Sender/files/13811094/messages_2.txt)

breiler, 2024-01-02:
I am not sure, but it looks like you are using the 32-bit version of UGS in these logs and you can very well hit the 2Gb memory limit, that is probably why the program becomes unresponsive. I would like a log file from the 64-bit edition, make sure that you have downloaded a 64-bit package:
https://universalgcodesender.com/download/

I noticed that it takes fairly long to load the model and that it contains some parsing warnings. Is it a big file? Do mind sharing one?

Username-CNC, 2024-01-02:
[raphael-3D Schlichten 1.zip](https://github.com/winder/Universal-G-Code-Sender/files/13811666/raphael-3D.Schlichten.1.zip)

breiler, 2024-01-02:
I had no problems loading the file but noticed that it requires quite alot memory - around 2.2Gb on my system. So make sure that you've downloaded the 64-bit packaging of UGS.

Username-CNC, 2024-01-02:
I will test everything again tomorrow with the 64 bit 2.1.3 version and send you the log file. Thank you for your quick message

Username-CNC, 2024-01-03:
I recently carried out the test again, unfortunately UGS freezes after a short time. See messages.log
I have the feeling that the cause is to be found in the spindle motor. Do you know this problem?
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/13821204/messages.log)
[messages_1.log](https://github.com/winder/Universal-G-Code-Sender/files/13821205/messages_1.log)

breiler, 2024-01-03:
Very strange, it still says that it can use maximum 2Gb.

Can you open the config file `C:\Program Files\ugsplatform-win\etc\ugsplatform.conf` and add the parameter `-J-Xmx4G`.

This should force a increase of the memory to 4Gb.

breiler, 2024-01-23:
Closing as we got no further response from user and I can't reproduce the problem.

closed 2024-01-23 by breiler