winder/Universal-G-Code-Sender #2417: UGS freeze after opening file, ver 2.1.0 to 2.1.4 and nightly

17dazer17, 2024-01-03:
### Version

2.1.4

### Hardware / Firmware

GRBL 1.1

### What happened

Can't  create new file, open file, can't open any tab in UGS. I can only conect to mashine and JOG. 

### How to reproduce

1.open UGS
2.connect
3.open file
4.no respond any tab, only able to JOG ns connect 
5 restart ugs
6. Step 1 to 5

I can open file from folder during UGS is running and stucked (click on '.nc' open with pick UGS then file is loading and I can run machine). After finish gcode, UGS is stucked(can JOG and connect, tabs not responding) . Then if i want open new gcode I go to folder, open with (as i wrote).

1.So running one g code
2.Restart UGS
3. Running another g code

Thats my UGS works. Tried all version nightly also. Few times works good for one/two days then UGS stucked. 

### Operating System

Win10 64bit pro

### Anything else

Almost every time, one restart one gcode, second restart second gcode

breiler, 2024-01-03:
Can you post your log file from:
C:/Users/[your username>]/AppData/Roaming/ugsplatform/var/log/messages.log

17dazer17, 2024-01-04:
[Uploading messages.log…]()

17dazer17, 2024-01-04:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/13827623/messages.log)

17dazer17, 2024-01-04:
Maybe I need to increase limit memory to 4gb?

breiler, 2024-01-04:
You are not loading a large file, so I don't think it is the issue with memory.

You are getting this error: `java.lang.InternalError: Bie??cy proces wykorzysta? wszystkie dozwolone przez system doj?cia do obiektów Window Manager.`

It roughly translate to `java.lang.InternalError: The current process uses all system-allowed handles to Window Manager objects.`

I have seen other reports that it may happen on multi monitor setups, do you have multiple monitors connected to your computer?

17dazer17, 2024-01-04:
This is the error translated

```
The current process has used all of its system allowance of handles for Window Manager objects
ERROR_NO_MORE_USER_HANDLES 1158
```

I olny use notebook with UGS, I have conection to local network drive (second notebook on which i małe gcode on sheetcam)

breiler, 2024-01-04:
I booted up my machine, I set the max memory to 2Gb and loaded a bunch of files.

I then opened the Task Manager and opened the Details tab and added the column for handles and this is what I got:
[image]

17dazer17, 2024-01-04:
[image]
This is mine

17dazer17, 2024-01-04:
https://github.com/winder/Universal-G-Code-Sender/assets/155579865/df511c1b-11b2-4126-b32d-f2e39f8e6db4

breiler, 2024-01-04:
Add the column for "Handles" in the Task manager.

breiler, 2024-01-04:
This might the a problem, are you using a network drive?
```
java.io.IOException: ReadDirectoryChangesW failed on Z:\: 53
	at org.netbeans.modules.masterfs.watcher.windows.WindowsNotifier.handleChanges(WindowsNotifier.java:406)
	at org.netbeans.modules.masterfs.watcher.windows.WindowsNotifier.access$200(WindowsNotifier.java:56)	
```

What happens if you move the file to the desktop and open it from there?

17dazer17, 2024-01-04:
Yes I am opening from network drive, I will move files to desktop, open files and run,then check after gcode finish

17dazer17, 2024-01-04:
Hi, gcode finished, UGS still stucked. I moved files to desktop and opened from it. I can still JOG
[image]

breiler, 2024-01-04:
Log file please

17dazer17, 2024-01-04:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/13834098/messages.log)

breiler, 2024-01-04:
This error was in your first log file also, I thought that it was symptom but it might be the cause. It seems to be a problem with the OpenGL integration, I have no idea what it is and I can't reproduce it. 

You could try to upgrade the graphics drivers, other than that I don't think I will be able to help you.

```
java.lang.reflect.InvocationTargetException
	at java.desktop/java.awt.EventQueue.invokeAndWait(Unknown Source)
INFO [null]: Last record repeated again.
	at java.desktop/javax.swing.SwingUtilities.invokeAndWait(Unknown Source)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:98)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:471)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:245)
	at java.base/java.util.TimerThread.mainLoop(Unknown Source)
	at java.base/java.util.TimerThread.run(Unknown Source)
Caused by: com.jogamp.opengl.GLException: Error making context 0x20000 current on Thread AWT-EventQueue-0, drawableWrite 0x4c010ced, drawableRead 0x4c010ced, werr: 0, WindowsWGLContext [Version 4.0 (Compat profile, arb, compat[ES2, ES3], FBO, hardware) - 4.0.0 - Build 10.18.10.4358 [GL 4.0.0, vendor 10.18.10 (- Build 10.18.10.4358)], options 0x4c03, this 0x73d9fa89, handle 0x20000, isShared false, jogamp.opengl.gl4.GL4bcImpl@167323a6,
	 quirks: [NoDoubleBufferedBitmap, NoSurfacelessCtx],
	Drawable: ResizeableImpl[Initialized true, realized true, texUnit 0, samples 0,
	Factory   jogamp.opengl.windows.wgl.WindowsWGLDrawableFactory@36021de9,
	Handle    0x4c010ced,
	Caps      GLCaps[wgl vid 7 arb: rgba 8/8/8/0, opaque, accum-rgba 0/0/0/0, dp/st/ms 16/0/0, one, mono  , hw, GLProfile[GL4bc/GL4bc.hw], offscr[fbo]],
	fboI back 0, front 0, num 1,
	FBO front read 1, FBO[name r/w 1/1, init true, bound false, size 818x868, samples 0/8, modified false/false, depth RenderAttachment[type DEPTH, format 0x81a5, samples 0, 818x868, name 0x1, obj 0x3d0c28eb], stencil null, colorbuffer attachments: 1/8, with 1 textures: [TextureAttachment[type COLOR_TEXTURE, target GL_TEXTURE_2D, level 0, format 0x8051, 818x868, border 0, dataFormat 0x1907, dataType 0x1401; min/mag 0x2600/0x2600, wrap S/T 0x812f/0x812f; name 0x1, obj 0x500566a7], null, null, null, null, null, null, null], msaa[null, hasSink false, dirty true], state OK, obj 0x6eedd169],
	FBO back  write 1, FBO[name r/w 1/1, init true, bound false, size 818x868, samples 0/8, modified false/false, depth RenderAttachment[type DEPTH, format 0x81a5, samples 0, 818x868, name 0x1, obj 0x3d0c28eb], stencil null, colorbuffer attachments: 1/8, with 1 textures: [TextureAttachment[type COLOR_TEXTURE, target GL_TEXTURE_2D, level 0, format 0x8051, 818x868, border 0, dataFormat 0x1907, dataType 0x1401; min/mag 0x2600/0x2600, wrap S/T 0x812f/0x812f; name 0x1, obj 0x500566a7], null, null, null, null, null, null, null], msaa[null, hasSink false, dirty true], state OK, obj 0x6eedd169],
	Surface   GDISurface[ displayHandle 0x0
, surfaceHandle 0x4c010ced
, size 818x868
, UOB[ OWNS_SURFACE | OWNS_DEVICE | WINDOW_INVISIBLE ]
, WindowsWGLGraphicsConfiguration[DefaultGraphicsScreen[WindowsGraphicsDevice[type .windows, connection decon, unitID 0, handle 0x0, owner false, NullToolkitLock[obj 0x49318170]], idx 0], pfdID 7, ARB-Choosen true,
	requested GLCaps[rgba 8/8/8/0, opaque, accum-rgba 0/0/0/0, dp/st/ms 16/0/0, one, mono  , hw, GLProfile[GL4bc/GL4bc.hw], on-scr[.]],
	chosen    GLCaps[wgl vid 7 arb: rgba 8/8/8/0, opaque, accum-rgba 0/0/0/0, dp/st/ms 16/0/0, one, mono  , hw, GLProfile[GL4bc/GL4bc.hw], offscr[fbo]]]
, surfaceLock <618ca101, 1a47c3c9>[count 1, qsz 0, owner <AWT-EventQueue-0>]
, GDIDummyUpstreamSurfaceHook[pixel 818x868]
, upstreamSurface false ]]] 
	at jogamp.opengl.windows.wgl.WindowsWGLContext.makeCurrentImpl(WindowsWGLContext.java:442)
	at jogamp.opengl.GLContextImpl.makeCurrentWithinLock(GLContextImpl.java:869)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:677)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:615)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1279)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.opengl.awt.GLJPanel$OffscreenBackend.doPaintComponent(GLJPanel.java:2127)
	at com.jogamp.opengl.awt.GLJPanel.paintComponent(GLJPanel.java:587)
	at java.desktop/javax.swing.JComponent.paint(Unknown Source)
	at java.desktop/javax.swing.JComponent.paintChildren(Unknown Source)
	at java.desktop/javax.swing.JComponent.paint(Unknown Source)
	at java.desktop/javax.swing.JComponent.paintChildren(Unknown Source)
	at java.desktop/javax.swing.JComponent.paint(Unknown Source)
	at java.desktop/javax.swing.JComponent.paintChildren(Unknown Source)
	at java.desktop/javax.swing.JComponent.paint(Unknown Source)
	at java.desktop/javax.swing.JComponent.paintChildren(Unknown Source)
	at java.desktop/javax.swing.JComponent.paint(Unknown Source)
	at org.netbeans.swing.tabcontrol.TabbedContainer.paint(TabbedContainer.java:975)
	at java.desktop/javax.swing.JComponent.paintChildren(Unknown Source)
	at java.desktop/javax.swing.JComponent.paint(Unknown Source)
	at java.desktop/javax.swing.JComponent.paintChildren(Unknown Source)
	at java.desktop/javax.swing.JComponent.paint(Unknown Source)
	at java.desktop/javax.swing.JComponent.paintChildren(Unknown Source)
	at java.desktop/javax.swing.JComponent.paint(Unknown Source)
	at java.desktop/javax.swing.JComponent.paintChildren(Unknown Source)
[... cut ...]
	at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBufferedImpl(Unknown Source)
	at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(Unknown Source)
	at java.desktop/javax.swing.RepaintManager$PaintManager.paint(Unknown Source)
	at java.desktop/javax.swing.RepaintManager.paint(Unknown Source)
	at java.desktop/javax.swing.JComponent._paintImmediately(Unknown Source)
	at java.desktop/javax.swing.JComponent.paintImmediately(Unknown Source)
	at java.desktop/javax.swing.RepaintManager$4.run(Unknown Source)
```

17dazer17, 2024-01-04:
I made few graphic updates, little bit more performance and update drivers, its better, I could open file few times after a lot of JOG but it now stopped, please one more time view my fresh LOG file and quote me that error, i will try to find out the problem on web

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/13835209/messages.log)

breiler, 2024-01-04:
The log file no longer contains OpenGL errors, instead you now get this:
```
java.lang.InternalError: The specified module could not be found.

	at java.desktop/sun.awt.windows.WToolkit.eventLoop(Native Method)
	at java.desktop/sun.awt.windows.WToolkit.run(Unknown Source)
[catch] at java.base/java.lang.Thread.run(Unknown Source)
```

I did a quick googling and couldn't find any obvious fix.

17dazer17, 2024-01-05:
I found this. Do upu think it could be helpfull? 

[https://forum.image.sc/t/custom-result-table-size-limit-window-manager-objects/44799/7](url)
[https://forum.image.sc/t/java-lang-internalerror-the-current-process-has-used-all-of-its-system-allowance-of-handles-for-window-manager-objects/45027](url)

breiler, 2024-01-05:
It no longer reports an error on the handles. Your screenshot of the task manager doesn't show an excessive use of handles either so I don't think it is a problem anymore.

This is the error you should be searching for:
`java.lang.InternalError: The specified module could not be found.`

17dazer17, 2024-01-05:
I reinstall to win7, all drivers are up to date. Problem with freezing may caused by not good graphic card. UGS is still stucked.
NVidia GT630M.
I switched to better card, not helped.
I need to learn how to live with this bug. I still can run gcode but I need reload it from folder not from UGS-open file.

breiler, 2024-01-05:
That sounds tedious, you should check out the alternative gcode senders:
https://github.com/winder/Universal-G-Code-Sender/wiki/Software-comparison#gcode-sender

17dazer17, 2024-01-06:
Yes. GT630 vram is 1gb, i switch to my third notebook (GT745 vram 4gb) and the problem gone. Everything works perfect. 

I have theory that Your visualiser and real time position takes a lot of memory. I found that sometimes I can see a frame (shadow) of tabs but I cant click, its not opening. 

Iam sure that is problem with vram memory. As I said on my better notebook problem is gone.

I like UGS, I am using joystick control and this is priority for me, I need something witch joystick /gamepad control. Is there any chance to turn off or delete from software the visualiser? Or mayby turn off 3d view or drop down the resolution? I am 99% sure that will help. To be honest I dont need live view and position , I always simulate track in sheetcam so I can live without visualiser but cant live without joystick. I was working without joystick, JOG clicking at notebook is not as comfortable as with joystick.

17dazer17, 2024-01-06:
[image]

breiler, 2024-01-06:
You can just close the visualizer and it shouldn't load OpenGL at all. 

We could also try to disable all visualizer features and enable one at a time to pinpoint if there is one particular that is causing the problem. Right click in the visualizer and go to **Show features** and disable them all. Then you can try and open a file and see if that works. Enable a new feature and open a file again, and so on...

I run Linux so things might work a bit different, but I loaded a 8Mb gcode file with 402276 lines and then ran "nvidia-smi":

```
joacim@kitty:~$ nvidia-smi 
Sat Jan  6 07:37:45 2024       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.147.05   Driver Version: 525.147.05   CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  N/A |
|  0%   45C    P8    21W / 170W |    601MiB / 12288MiB |     14%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1974      G   /usr/lib/xorg/Xorg                375MiB |
|    0   N/A  N/A      2117      G   /usr/bin/gnome-shell               98MiB |
|    0   N/A  N/A     73982      G   /usr/bin/nvidia-settings            2MiB |
|    0   N/A  N/A     75373    C+G   ...365983298466638326,262144       89MiB |
|    0   N/A  N/A     75874      G   ...temurin-17.0.8.1/bin/java       31MiB |
+-----------------------------------------------------------------------------+
```

It says that UGS is using 31Mb of video memory.

breiler, 2024-01-06:
If you get fed up with UGS, I checked the gSender documentation and they support gamepads as well: https://resources.sienci.com/view/gs-additional-features/

17dazer17, 2024-01-06:
Thank You. In card details on my better notebook dedicated video memory is set to 128mb. 

On second where problem occurs dedicated video memory is 32mb.. I will try to increase that parameters and then try your method ;)

andrewmurraydavid, 2024-02-10:
@17dazer17 were you using the line follower? Ive had similar issues on Linux (my CNC control tower is an old scrappy office PC with Ubuntu on it) and when running programs with over 4000-5000 lines and the line follower on, it gets stuck and I either have to wait until UGS "finishes" running the program in memory or just exit UGS.
Even though your program has about 1350 lines, you have many arcs, which my guess is that it adds up to over 5000 after the arcs are "converted" into segments.
Again, all that would be the case if you have the line follower on in the editor.

breiler, 2024-03-14:
Can see from the log that you had a gamepad activated. The library we were using had a memory leak which is now fixed in version 2.1.6 (#2485)

closed 2024-03-14 by breiler