winder/Universal-G-Code-Sender #2418: Trace Image Dialog

wbrogdo1, 2024-01-07:
### Version

2.1.4

### Hardware / Firmware

GRBL 1.1

### What happened

Clicking on Trace image dialog doesn't open any window for selecting an image

### How to reproduce

click on trace image dialog

### Operating System

windows 11 pro

### Anything else

it doesn't open anytime i select it

breiler, 2024-01-07:
Thanks, it seems to be a problem with the Java range slider.

I am getting this error:
```
java.lang.NullPointerException: Cannot invoke "java.lang.Integer.intValue()" because the return value of "javax.swing.UIManager.get(Object)" is null
	at java.desktop/javax.swing.plaf.metal.MetalSliderUI.installUI(MetalSliderUI.java:154)
	at java.desktop/javax.swing.JComponent.setUI(JComponent.java:730)
	at com.jidesoft.swing.RangeSlider.updateUI(RangeSlider.java:90)
	at java.desktop/javax.swing.JSlider.<init>(JSlider.java:308)
	at com.jidesoft.swing.RangeSlider.<init>(RangeSlider.java:68)
	at com.willwinder.ugs.nbp.designer.gui.imagetracer.TraceSettingsPanel.<init>(TraceSettingsPanel.java:41)
	at com.willwinder.ugs.nbp.designer.gui.imagetracer.ImageTracerDialog.<init>(ImageTracerDialog.java:46)
	at com.willwinder.ugs.nbp.designer.actions.TraceImageAction.actionPerformed(TraceImageAction.java:62)
```

This is the line failing for me in `javax.swing.plaf.metal.MetalSliderUI`:
```
tickLength = safeLength = ((Integer)UIManager.get( "Slider.majorTickLength" )).intValue();
```

closed 2024-01-14 by breiler

breiler, 2024-01-14:
Yeah sorry, it auto closed the issue when merging.

It is available in the latest brightly build if you want to try it out. Or you will have to wait until version 2.1.5.

https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

PR #2424 (2024-01-14): Add missing dependency for jide-oss with flatlaf
Fixes #2418