winder/Universal-G-Code-Sender #2419: Probe and autolevel fail to change Z zero

Acliad, 2024-01-07:
### Version

2.1.3

### Hardware / Firmware

TinyG

### What happened

When running the Z probe module, my CNC appears to correctly interrupt the `G38.2 Z-20 F100` command and lowers Z until the touch plate is hit, at which point it stops. However, UGS does nothing more after that. The DRO reads the Z position of the touch plate and does not zero out the axis. In auto leveling mode, the same thing happens and the routine does nothing more after the first probe (i.e., autoleveling fails).

I'm on the latest TinyG Firmware (0.97, 440.20). 

### How to reproduce

Run "Probe and zero Z". 

I'm not sure if this is a TinyG specific issue, as that's the only controller I have. 

### Operating System

Raspbian GNU/Linux 11 (bullseye)

### Anything else

This is the output of the console from hitting "Probe and zero Z" to it stopping:

```
>>> G10 L20 P1 Z0
>>> G90 G21
{"r":{},"f":[1,0,14,73]}
>>> G21 G91 G49
{"r":{},"f":[1,0,8,4401]}
>>> G38.2 Z-20 F100
{"r":{},"f":[1,0,12,71]}
{"sr":{"dist":1}}
>>> G90 G21
{"r":{},"f":[1,0,16,75]}
{"sr":{"posy":0.000,"stat":7,"dist":0,"coor":0}}
{"sr":{"posz":-0.290,"mpoz":-0.298,"vel":100.00}}
{"sr":{"posz":-0.615,"mpoz":-0.615}}
{"sr":{"posz":-0.940,"mpoz":-0.940}}
{"sr":{"posz":-1.265,"mpoz":-1.265}}
{"sr":{"posz":-1.590,"mpoz":-1.590}}
{"sr":{"posz":-1.863,"mpoz":-1.863,"vel":0.22}}
{"r":{"prb":{"e":1,"x":0.000,"y":0.000,"z":-1.863,"a":0.000,"b":0.000,"c":0.000}},"f":[1,0,0,847]}
{"sr":{"posy":-200.000,"vel":0.00,"stat":3,"dist":1,"coor":1}}
{"r":{},"f":[1,0,8,4401]}
{"sr":{"dist":0}}
```
Probe settings:
[image]

State after the probe touches:
[image]

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/13855281/messages.log)

wbrogdo1, 2024-01-08:
Mine doesn’t work that way either. However, after running the z probe plug-in, z axis is zeroed as long as I don’t click the zero button in the controller state window

Acliad, 2024-01-09:
I'm not quite sure what you mean. When you run the z probe plug-in, it **does** zero out Z for you? Does it respect the "Touch plate thickness" setting? Does the AutoLeveler plug-in work for you?

wbrogdo1, 2024-01-09:
when I position my x & y axis, I click the zero button in the controller state window. the window updates to show zero.(just like your picture)
when I run the z probe module, it zero's it's position but doesn't reflect a zero in the controller state window. The z axis is in fact zeroed as long as I don't click the zero button in the controller state window that's directly below the idle or run wording.

After clicking probe and zero z, just click return to zero in your toolbox and see if your z axis is actually zeroed.

Acliad, 2024-01-09:
Oh, I think I see. If I run the Z Probe and then hit return to zero (Or type G0 Z0), mine returns to the previous zero position, it doesn't go to the zero position of the touch plate. The routine is definitely failing too, as I believe it's supposed to raise a little after touching the plate, and the auto leveler only does a single probe.

kbrown73, 2024-02-08:
I'm experiencing the same. Probe goes down and on contact it just stops and doesn't appear to do anything else. I'm running latest stable g2core on Arduino Due + [g2core shield](https://webshop.djuke.nl/kits/g2shield-onboard-kit)

Acliad, 2024-02-08:
Ahh, interesting. I actually didn't expect this to happen on GRBL. I don't get the unhandled message with tinyg. What OS are you running? I've also tried this on Debian with an Intel machine and have the same issue as the Pi.

kbrown73, 2024-02-08:
I messed about with it a bit more and it turns out that unhandled message was linked to a setting I had compiled g2core with:

```
#ifndef PROBE_REPORT_ENABLE
#define PROBE_REPORT_ENABLE         true    // {prbr:
#endif
```

I tried with it set to false and it indeed got rid of that unhandled message line, but it did not solve the original issue though. It still just stopped on probe contact.