winder/Universal-G-Code-Sender #2439: no buttun enabled when hold using m0

megahacker91, 2024-01-26:
### Version

2.1.4

### Hardware / Firmware

FluidNC

### What happened

i want to use manual tool change in my gcode i use m0 at start of new tool
when it pause (hold) i can not jog or run z probe macro 

### How to reproduce

it should not disable all buttun and macros when m0 code send so ican jog and probe

### Operating System

linux mint 21.3

breiler, 2024-01-26:
That is how `M0` is supposed to work, the controller is put in a `HOLD` state and will not accept any other commands than "stop" or "resume".

The command you are looking for is `M6` which UGS currently doesn't support.

I am closing this as as a duplicate since we have other requests for supporting tool change:
https://github.com/search?q=repo%3Awinder%2FUniversal-G-Code-Sender+%22tool+change%22&type=issues

closed 2024-01-26 by breiler

megahacker91, 2024-01-26:
i was using mach3 i was able to jog  and probe it after m0 command

megahacker91, 2024-01-26:
i had modefied the post processor to fix m6 problem.
insted of m6 the gcode go to 10mm on z 
then go to 0x 0y and wait resume after i put next  tool  using m0.
g0 z10
g0x0y0
m0

but the problem now is after m0 i can not probe the new tool

breiler, 2024-01-26:
> i was using mach3 i was able to jog and probe it after m0 command

As mentioned earlier, `M0` will set GRBL-based controllers in a feed hold state, no commands will be allowed. Much of the GRBL based controllers are based on the LinuxCNC spec which seems to have the same behavior: https://linuxcnc.org/docs/devel/html/gcode/m-code.html#mcode:m0-m1

UGS could probably implement some hacks to allow jogging by creating it's own virtual state which would intercept `M0`. But I think this would be a bad idea.

Just as an example, consider this gcode program where we move to a absolute position X10, Y10, Z10. Then we do a relative movement to X20. After that it makes the controller enter a feed hold with M0. The next command should do yet another relative movement and place the machine at X30. But if you have jogged somewhere during the M0 you will end up +X10 from your current jogged posistion.
```gcode
G90 G0 X10 Y10 Z10
G91 G1 X10 F1000
M0
G91 G1 X10
```

It's better to focus on implementing real support for a `M6` command instead of hacking together a custom `M0` behavior.