winder/Universal-G-Code-Sender #2440: Start button high risk of miss clicking it

villamvadasz, 2024-01-27:
### Description

Hi,

I have the problem, that sometimes I missclick the start button. It is happening not so often, but still when it happens it is annoying. Usually I want to load an other file, and somehow I click the start button. Then the machine starts to mill.

For this I could imagine a new feature, where next to the start button a check box is there, which have to be marke, in order that the start button becomes enabled. Once the milling is finisched (G code M2 to detect maybe?) the start button should become again unavailable. But during the milling the start button should stay available. This way the missclicking of the button could be reduced.

Not sure if others have also the problem. So I think it is fine if you just reject this issue if you think it is not worth to implement such a feature.

Thank You!

breiler, 2024-01-28:
Thanks for the feedback.

Similar requests has been made before with the suggestion to add a popup asking if the user really wants to run the program. I would probably go crazy having to confirm every time I started a job, so I would probably want a checkbox in the confirmation popup to never ask again.

I am also interested to learn why you feel that the play button should open a new file. For instance is there another software you have been using that behaves like this?

villamvadasz, 2024-01-29:
> I would probably want a checkbox in the confirmation popup to never ask again.
Sounds like a perfect solution!

> I am also interested to learn why you feel that the play button should open a new file. For instance is there another software you have been using that behaves like this?

I do not expect that the play button should open a new file. It is just misclicking. I just click the wrong button. I think, it is because somehow mentally the play button looks more dominant (like it does more thing) than the file open button, and this might divert my attention to it.