winder/Universal-G-Code-Sender #2449: Dialog Boxes not showing up

N0TB0T, 2024-02-02:
### Version

2.1.4

### Hardware / Firmware

GRBL 1.1

### What happened

The dialog boxes for no matter what ("Open File", "Options", ...) won't open. Only immediately after a restart, there's a guarantee they will, but after some clicks here and there, menu items will turn blue/selected when clicked, but nothing happens.

### How to reproduce

Can reproduce behaviour any time when running UGS.

### Operating System

Windows 10, nothing special at all.

breiler, 2024-02-02:
Please attach your log file from 
C:/Users/[your username>]/AppData/Roaming/ugsplatform/var/log/messages.log

N0TB0T, 2024-02-02:
So I guess it comes down to:

"java.lang.InternalError: The current process has used all of its system allowance of handles for Window Manager objects"

?

breiler, 2024-02-07:
Sorry for the late reply. 

I have no idea what is causing this error. Similar message has occurred for this user:
https://github.com/winder/Universal-G-Code-Sender/issues/2417

I am wondering if dropbox is doing strange things with the files? Maybe try placing UGS in another directory, `C:\Users\buddybrand` for instance?

N0TB0T, 2024-02-07:
There might be some flagging/locking going on in the background, i don't know either. Since moving the files to another location (C:) everything's smooth.

breiler, 2024-03-14:
This was likely caused by this issue #2465, fixed in UGS v2.1.6

closed 2024-03-14 by breiler