winder/Universal-G-Code-Sender #2450: Project Location from Carveco to UGS

Lemiel-Priff, 2024-02-03:
### Version

Other

### Hardware / Firmware

GRBL 1.1

### What happened

Hello,
I am currently designing in Carveco, working in millimeters, I am centering all of my pads, and am positioning my project at the bottom left pixel. I am importing with the machine file format of UGS GRBL mm .Gcode. When I open the project in UGS, the project is wedged to the right part of the work area, thereby cutting off the right side of my project. I am admittedly relatively new to CNCing, and would appreciate your assistance. Thank you in advance.

### Operating System

Windows 11 Pro, version 22 H2

### Anything else

This is happening every time I move from Carveco to UGS.

breiler, 2024-02-03:
Can you post a sample file?

Lemiel-Priff, 2024-02-03:
Yes, thank you for your assistance. 

T1
G17
G21
G90
G0Z20.7500
G0X0.0000Y0.0000
S22000M3
G0X34.9172Y30.2616Z2.0000
G1Z-0.7840F360.0
G1X61.3328Y30.2616F600.0
Y39.8014
X34.9172Y39.8014
Y49.3411
X61.3328Y49.3411
Y58.8808
X34.9172Y58.8808
Y68.4205
X61.3328
Y77.9603
X34.9172Y77.9603
Y87.5000
X61.3328
X61.3328Y97.0397
X34.9172
Y106.5795
X61.3328
Y116.1192
X34.9172Y116.1192
Y125.6589
X61.3328Y125.6589
Y135.1986
X34.9172
Y144.7384
X61.3328
X34.9172
Y30.2616
X61.3328Y30.2616
X61.3328Y144.7384
G0Z2.0000
G0X34.9172Y30.2616
G1Z-1.5680F360.0
G1X61.3328Y30.2616F600.0
Y39.8014
X34.9172Y39.8014
Y49.3411
X61.3328Y49.3411
Y58.8808
X34.9172Y58.8808
Y68.4205
X61.3328
Y77.9603
X34.9172Y77.9603
Y87.5000
X61.3328
X61.3328Y97.0397
X34.9172
Y106.5795
X61.3328
Y116.1192
X34.9172Y116.1192
Y125.6589
X61.3328Y125.6589
Y135.1986
X34.9172
Y144.7384
X61.3328
X34.9172
Y30.2616
[... 718 lines cut ...]
Y77.9603
X34.9172Y77.9603
Y87.5000
X61.3328
X61.3328Y97.0397
X34.9172
Y106.5795
X61.3328
Y116.1192
X34.9172Y116.1192
Y125.6589
X61.3328Y125.6589
Y135.1986
X34.9172
Y144.7384
X61.3328
X34.9172
Y30.2616
X61.3328Y30.2616
X61.3328Y144.7384
G0Z2.0000
M5
G0Z20.7500
G0X0.0000Y0.0000
M2

breiler, 2024-02-04:
Not sure that I understand, the project looks normal to me:
[image]

If you are expecting the lower left corner of the shape should be put in the origo (X0, Y0) this could just mean that you have some offsets set in your Carveco project.

breiler, 2024-03-14:
Closing as we got no further information from user