winder/Universal-G-Code-Sender #2452: Connection trouble

dastultz, 2024-02-06:
### Version

2.1.2

### Hardware / Firmware

GRBL 1.1

### What happened

Years ago I ran UGS with a different computer. I'm just getting started again with a different computer, the third I've used to run UGS. I have a consistent problem connecting. Upon connection attempt I get this:

*** Connecting to jserialcomm://ttyUSB0:115200
*** Fetching device status
>>> ?
*** Fetching device status (2 of 3)...
>>> ?
*** Fetching device status (3 of 3)...
>>> ?
**** Connection closed ****

Then I connect with a "serial monitor" program, in my case "screen" on Linux. I connect successfully, noting that it takes longer to connect than it seems UGS is waiting.
I exit "screen" and connect with UGS and get this (with verbose logging on):

*** Connecting to jserialcomm://ttyUSB0:115200
*** Fetching device status
>>> ?
<Idle|MPos:0.0000,0.0000,0.0000|F:0.0|WCO:-0.3630,-0.0149,-1.2425>
ok
>>> $I
[VER:1.1f.20170801:]
[OPT:S,15,128]
ok
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 1    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 1    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 50.000    (Homing locate feed rate, mm/min)
$25 = 457.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 2800    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 314.961    (X-axis travel resolution, step/mm)
$101 = 314.961    (Y-axis travel resolution, step/mm)
$102 = 314.961    (Z-axis travel resolution, step/mm)
$110 = 750.000    (X-axis maximum rate, mm/min)
$111 = 750.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 15.000    (X-axis acceleration, mm/sec^2)
$121 = 15.000    (Y-axis acceleration, mm/sec^2)
$122 = 15.000    (Z-axis acceleration, mm/sec^2)
$130 = 225.000    (X-axis maximum travel, millimeters)
$131 = 125.000    (Y-axis maximum travel, millimeters)
$132 = 170.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G20 G90 G94 M5 M9 T0 F0.0]
ok
*** Connected to GRBL 1.1f
<Idle|MPos:0.0000,0.0000,0.0000|F:0.0|Ov:100,100,100|A:C>
<Idle|MPos:0.0000,0.0000,0.0000|F:0.0>
<Idle|MPos:0.0000,0.0000,0.0000|F:0.0>
<Idle|MPos:0.0000,0.0000,0.0000|F:0.0>
<Idle|MPos:0.0000,0.0000,0.0000|F:0.0>
<Idle|MPos:0.0000,0.0000,0.0000|F:0.0>

I should note that I am running a customized version of GRBL but I haven't made any changes in 7 years and I've run it with UGS on 2 different computers.

I'm now running Fedora on a Framework 13 laptop. More details in the attached log file. Let me know if I can give you more information. Thanks for taking a look.

/Daryl

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/14173738/messages.log)

### How to reproduce

Try to connect with Framework 13 AMD laptop running Fedora.

### Operating System

Fedora

breiler, 2024-02-06:
Can you try the version 2.1.4, it is using the latest JSerialComm library which has some changes to the RTS/DTR functions. I have never experienced these kinds of problems on my workstation though (Ubuntu 23.04) so I have no idea if this will have any effect.

I am curious what do you mean by this?
> I connect successfully, noting that it takes longer to connect than it seems UGS is waiting.

UGS will try and send blank lines upon connect to make sure the controller is responsive (not in a terminal Alarm state which requires a reset). It will try three times with one second delay. If this takes longer on your hardware I'm interested what type of controller you are running.

dastultz, 2024-02-06:
2.1.4 behaves the same.

You got what I meant about timing. UGS tries for 3 seconds to connect. I count in my head 5 or 6 seconds for `screen` to connect. After that UGS connects the first try. So maybe the hardware has some kind of initialization delay? Again it's a Framework 13 AMD laptop. I don't know much about modern serial hardware, but I can try to find more information if you can suggest a query.

Here's a bunch of info the OS gave me:

pcilib: Error reading /sys/bus/pci/devices/0000:00:08.3/label: Operation not permitted
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 14e8
	Subsystem: Framework Computer Inc. Device 0006
	Flags: fast devsel

00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Device 14e9
	Subsystem: Framework Computer Inc. Device 0006
	Flags: fast devsel, IRQ 29
	Capabilities: <access denied>

00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 14ea
	Flags: fast devsel, IOMMU group 0

00:02.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 14ea
	Flags: fast devsel, IOMMU group 1

00:02.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 14ee (prog-if 00 [Normal decode])
	Subsystem: Advanced Micro Devices, Inc. [AMD] Device 1453
	Flags: bus master, fast devsel, latency 0, IRQ 38, IOMMU group 1
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: [disabled] [32-bit]
	Memory behind bridge: 90b00000-90bfffff [size=1M] [32-bit]
	Prefetchable memory behind bridge: 8010900000-80109fffff [size=1M] [32-bit]
	Capabilities: <access denied>
	Kernel driver in use: pcieport

00:02.4 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 14ee (prog-if 00 [Normal decode])
	Subsystem: Advanced Micro Devices, Inc. [AMD] Device 1453
	Flags: bus master, fast devsel, latency 0, IRQ 39, IOMMU group 1
	Bus: primary=00, secondary=02, subordinate=02, sec-latency=0
	I/O behind bridge: [disabled] [32-bit]
	Memory behind bridge: 90a00000-90afffff [size=1M] [32-bit]
	Prefetchable memory behind bridge: [disabled] [64-bit]
	Capabilities: <access denied>
	Kernel driver in use: pcieport

00:03.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 14ea
	Flags: fast devsel, IOMMU group 2

00:03.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 19h USB4/Thunderbolt PCIe tunnel (prog-if 00 [Normal decode])
	Subsystem: Advanced Micro Devices, Inc. [AMD] Device 1453
	Flags: bus master, fast devsel, latency 0, IRQ 40, IOMMU group 2
	Bus: primary=00, secondary=03, subordinate=61, sec-latency=0
	I/O behind bridge: 6000-9fff [size=16K] [16-bit]
	Memory behind bridge: 78000000-8fffffff [size=384M] [32-bit]
	Prefetchable memory behind bridge: 7000000000-7fffffffff [size=64G] [32-bit]
	Capabilities: <access denied>
	Kernel driver in use: pcieport

00:04.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 14ea
	Flags: fast devsel, IOMMU group 3

00:04.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 19h USB4/Thunderbolt PCIe tunnel (prog-if 00 [Normal decode])
	Subsystem: Advanced Micro Devices, Inc. [AMD] Device 1453
	Flags: bus master, fast devsel, latency 0, IRQ 41, IOMMU group 3
	Bus: primary=00, secondary=62, subordinate=c0, sec-latency=0
	I/O behind bridge: 2000-5fff [size=16K] [16-bit]
	Memory behind bridge: 60000000-77ffffff [size=384M] [32-bit]
	Prefetchable memory behind bridge: 6000000000-6fffffffff [size=64G] [32-bit]
	Capabilities: <access denied>
	Kernel driver in use: pcieport

00:08.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 14ea
	Flags: fast devsel, IOMMU group 4
[... 166 lines cut ...]
	Capabilities: <access denied>
	Kernel driver in use: xhci_hcd

c3:00.4 USB controller: Advanced Micro Devices, Inc. [AMD] Device 15c1 (prog-if 30 [XHCI])
	Subsystem: Framework Computer Inc. Device 0006
	Flags: bus master, fast devsel, latency 0, IRQ 52, IOMMU group 4
	Memory at 90700000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: <access denied>
	Kernel driver in use: xhci_hcd

c3:00.5 USB controller: Advanced Micro Devices, Inc. [AMD] Pink Sardine USB4/Thunderbolt NHI controller #1 (prog-if 40 [USB4 Host Interface])
	Subsystem: Framework Computer Inc. Device 0006
	Flags: bus master, fast devsel, latency 0, IRQ 78, IOMMU group 4
	Memory at 90800000 (64-bit, non-prefetchable) [size=512K]
	Capabilities: <access denied>
	Kernel driver in use: thunderbolt
	Kernel modules: thunderbolt

c3:00.6 USB controller: Advanced Micro Devices, Inc. [AMD] Pink Sardine USB4/Thunderbolt NHI controller #2 (prog-if 40 [USB4 Host Interface])
	Subsystem: Framework Computer Inc. Device 0006
	Flags: bus master, fast devsel, latency 0, IRQ 95, IOMMU group 4
	Memory at 90880000 (64-bit, non-prefetchable) [size=512K]
	Capabilities: <access denied>
	Kernel driver in use: thunderbolt
	Kernel modules: thunderbolt

breiler, 2024-02-07:
I've made a small change in the latest nightly build so that it attempts the connection for ten seconds. I have no idea if this will work since it will not be waiting for a response from the controller instead it will "ping" it with newlines to see if it is alive. This may disrupt its boot sequence which I had problems with on Arduino Megas.

Regarding what controller you are using, I could not find it in your list. What kind of device is it? Arduino Due, Arduino Nano, ESP32, ...?

dastultz, 2024-02-07:
Sorry I misunderstood the question about the controller. I thought we were talking about the USB/serial hardware on my computer. GRBL is running on a Sanguino.

I can't figure out how to access the nightly build asset.

breiler, 2024-02-08:
The column that says "Nightly build":
https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

dastultz, 2024-02-09:
That did the trick! Here's the output:

```
*** Connecting to jserialcomm://ttyUSB0:115200
*** Fetching device status
>>> ?
*** Fetching device status (2 of 10)...
>>> ?
*** Fetching device status (3 of 10)...
>>> ?
*** Fetching device status (4 of 10)...
>>> ?
*** Fetching device status (5 of 10)...
>>> ?
**** The communicator has been paused ****
**** The communicator has been paused ****
Grbl 1.1f ['$' for help]
>G90G20:ok
<Idle|MPos:0.0000,0.0000,0.0000|F:0.0|WCO:-0.3630,-0.0149,-1.2425>
*** Fetching device status (6 of 10)...
>>> ?
<Idle|MPos:0.0000,0.0000,0.0000|F:0.0|Ov:100,100,100|A:C>
ok
>>> $I
[VER:1.1f.20170801:]
[OPT:S,15,128]
ok
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 1    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 1    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 50.000    (Homing locate feed rate, mm/min)
$25 = 457.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 2800    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 314.961    (X-axis travel resolution, step/mm)
$101 = 314.961    (Y-axis travel resolution, step/mm)
$102 = 314.961    (Z-axis travel resolution, step/mm)
$110 = 750.000    (X-axis maximum rate, mm/min)
$111 = 750.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 15.000    (X-axis acceleration, mm/sec^2)
$121 = 15.000    (Y-axis acceleration, mm/sec^2)
$122 = 15.000    (Z-axis acceleration, mm/sec^2)
$130 = 225.000    (X-axis maximum travel, millimeters)
$131 = 125.000    (Y-axis maximum travel, millimeters)
$132 = 170.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G20 G90 G94 M5 M9 T0 F0.0]
ok
*** Connected to GRBL 1.1f
**** Connection closed ****
```

closed 2024-02-09 by dastultz