winder/Universal-G-Code-Sender #2465: Problem to Open File after run one gocode Windows 11

3hrsolucoes, 2024-02-15:
### Version

2.1.4

### Hardware / Firmware

Grbl_ESP32

### What happened

So, I have a problem to open file after run one gcode. Its need to be closed and re open the UGS. I try reisntall the windows but nothing. I read one problem here with folder that contains other files, my folder of gcode have 3 files (.nc). the folder of UGS and folder Gcode are in C:/

The older version, i dont remember which one, doesnt had this issue.

Thanks

### Operating System

Windows 11

breiler, 2024-02-15:
Can you share your log file:
C:/Users/[your username>]/AppData/Roaming/ugsplatform/var/log/messages.log

You can also open the log directory from the menu Tools \ Open log directory

3hrsolucoes, 2024-02-15:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/14300000/messages.log)

Thanks

breiler, 2024-02-15:
No errors in log and it looks like you are able to select the file multiple times.

I have replaced the file open dialog in the latest version of UGS 2.1.5, you could try and upgrade: https://universalgcodesender.com/download/

3hrsolucoes, 2024-02-15:
Omg, sorry i'm using 2.1.5

breiler, 2024-02-15:
Oh right, it's in the log, I didn't bother checking as you wrote it in the issue description. =)

I tried rebooting into Windows 10 and tried with with UGS 2.1.5 and I can't reproduce the problem. Do you have an Antivirus program or something that may prevent UGS from opening the file dialog?

3hrsolucoes, 2024-02-15:
Ok, no, i complete reinstall the windows 11 today, only the UGS and Winrar are installed now.

3hrsolucoes, 2024-02-15:
i'm testing version 2.1.4 and not problems until now ! maybe new version 2.1.5 have problems with windows 11

breiler, 2024-02-16:
I tried upgrading to Windows 11 and I can still not reproduce it...

Worst case we'll have to create a wrapper for the file chooser to use the current AWT-based on MacOSX and the Swing-based on Linux and Windows.

If anyone else has the same problem please attach log files.

3hrsolucoes, 2024-02-16:
Ok, i'm testing on UGS 2.1.4 and have same problem today.

:/

3hrsolucoes, 2024-02-16:
maybe the problem is on my computer, the computer is mini computer intel from china.

3hrsolucoes, 2024-02-22:
I Think i have solved the problem..  I tested this and ultil now its OK ! Version UGS 2.1.4

https://login.microsoftonline.com/common/oauth2/v2.0/authorize?client_id=a81d90ac-aa75-4cf8-b14c-58bf348528fe&redirect_uri=https%3A%2F%2Fanswers.microsoft.com&response_type=code%20id_token&scope=openid%20profile&state=OpenIdConnect.AuthenticationProperties%3DWfw7fPfrjEuo5mdsv91i1_GRHTkxN-IHS0sffyKLxHUTEb0wAi5IEsEhGPF9NzDC6XebDZMn5uL_-1vJLAd95MZJTETps3jjyaVe4QcGIAIsbNceMI342KyNlOXlRb2kbKd_23jv9qBZc-2p5DUpTF-1aAwvvYBcruFqden3tuXTXGzyBPqeT3c5KBv48fCiosHIFtCNq42FX4eyM3Z-U2B5BIvcL_UYYp2nQBCDdhx5_xfDXGMSYezwGpOs7Ys21n4YwVyojRZO-0kOSI3BoeTflBpV_oezaNn0yxg-2HvRBMMxQ_htx0xtoNyZp4i39upkoa_6Vn-2nw569I-lsEEd5L28AuYtpQR-SiBcTUSvFONQ2QYY5wqnMr8MR3Qd&response_mode=form_post&nonce=638442419708514226.MzliN2I3ZTktZDhkZi00MGFhLTg0YjEtZTc0OGFiMjFmM2I2YmE4NWNmNTAtYmRlMy00NDNhLTkzZDUtODU0MGMyZWZmNjc0&nopa=2&prompt=none&x-client-SKU=ID_NET472&x-client-ver=6.17.0.0

3hrsolucoes, 2024-02-22:
I think i know the problem ! I tested this today and until now everything is OK.

https://answers.microsoft.com/pt-br/windows/forum/all/erro-do-sistema-c%c3%b3digo-1158-o-processo-atual/69476f95-4d17-4db5-b475-38477f5f93dc

SaihbJ, 2024-02-27:
Hello !
I have the same problem, and I tried the last solution (with a bad Italian (Spanish ?) translation) and now my computer doesn't even start..!! I'm good for a windows reinitialization I guess.
But anyway I had the same problem, I was able to use UGS 5 minutes and then impossible to open a new project. I'm using a joystick to control my CNC machine and I wonder if this has a link...

3hrsolucoes, 2024-02-27:
Hello ! try not do the last step !.. I have a problem with the final step and forget to post here.

I edited my last comment. Try what is there now !

breiler, 2024-02-29:
Thanks @SaihbJ, this does indeed seem to be related to the gamepad library.

I have posted a new issue here: https://github.com/libgdx/Jamepad/issues/28

SaihbJ, 2024-02-29:
Ok, thank you @breiler ! I have a fresh new Windows 11 installed but haven't tried UGS yet... I'd like to be able to use my joystick so let's hope that it will be fixed !

SaihbJ, 2024-02-29:
I actually also had some strange java errors : "a java.lang.nullpointerexception exception has occurred"
Can it be linked ?

breiler, 2024-03-08:
I have created a new version here which fixed it on my computer: https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.1.6

I noticed that the Y and Z axis was inverted with my controller so I needed to invert those axes in the settings. If this is because a changed gamepad mapping in the controller database or if something changed in the library I don't know. Since it is easy to invert the axes I didn't investigate it further.
[image]

SaihbJ, 2024-03-08:
Thank you @breiler !
I'll test it today and keep you in touch.

breiler, 2024-03-14:
This was likely caused by a bug in the gamepad library we were using. Fixed in #2478, available in UGS version 2.1.6.

closed 2024-03-14 by breiler