winder/Universal-G-Code-Sender #2479: UGS does not open

Helhagg, 2024-03-08:
### Version

2.1.5

### Hardware / Firmware

GRBL 1.1

### What happened

I have been using UGS for 6 weeks now and this is the second time it has happened.  the icon appears on my screen, looks like it's loading, the progress bar gets to the end and the icon disappears but the program does not open.  Last time I deleted and reinstalled and it was fine.  This time I have tried shutting down my laptop, deleting and reinstalling UGS but it continues to behave the same way.  I was able to get my CNC up and running using the usb stick provided by my cnc manufacturer that includes a version of UGS on it but I prefer to work with the program directly from my laptop.  

### How to reproduce

I have provided this information above, however, when I try to open the file, it appears to open, the UGS icon appears on my screen with the progress bar indicating it is loading.  When the progress bar is done, the icon disappears and the program never loads.  I have tried running as administrator as well as just opening the file but neither option works.  

### Operating System

Windows 11

### Anything else

Every time I try to run the program.  I cannot get it to open and have had to resort to running it from the usb stick provided by my CNC manufacturer in order to keep production moving.

breiler, 2024-03-09:
Can you post the log file from, it may give some clues what happened: 
`C:/Users/[your username>]/AppData/Roaming/ugsplatform/var/log/messages.log`

I suspect that the configuration has gone corrupt and the quickest way to get it sorted is to delete the settings folder: `C:/Users/[your username>]/AppData/Roaming/ugsplatform`

breiler, 2024-03-15:
Closing as we did not get any response from user

closed 2024-03-15 by breiler