winder/Universal-G-Code-Sender #2480: Communication issue between UGS and my CNC

Helhagg, 2024-03-08:
### Version

2.1.5

### Hardware / Firmware

GRBL 1.1

### What happened

Yesterday in the middle of cutting a file on my CNC, the CNC just stopped moving and my UGS registered a status of IDLE.  This has never happened before and I have been using both UGS and my CNC together for the past 6 weeks pretty much nonstop for 7-10 hours a day, 7 days a week as I try to get my small business off the ground.  Today the exact same thing happened.  I was cutting a file and the CNC just stopped moving and UGS indicated the CNC was IDLE.  I was able to choose a location in my file to run from and was able to save my piece but that's twice in two days and I can't keep having this happen.  

### How to reproduce

Again I have indicated this above.  I have run files on two separate occasions on my CNC, two days in a row, and both times UGS appears to have just stopped communicating with my CNC leaving it in an idle state in the middle of the file.  

### Operating System

windows 11

### Anything else

Twice in two days now.  As I cut my next two files I will submit again with any updates but I sure would like to know what's going on.

breiler, 2024-03-09:
Can you post the last couple of messages from the console when this happens. 

What is your hardware setup?

A common problem is that CNC controllers restarts due to EMI. And it is especially common problem with cheaper controllers that doesn't use filtering and/or isolation on inputs. 

Make sure that inputs wires and the USB cable is placed away from high power cables such as spindle power. If you are using a dust collector, make sure that it is grounded as the hose can build up a lot of static electricity. 

Do not use any USB cable extension or USB hub.

breiler, 2024-03-15:
Closing as we did not get any response from user

closed 2024-03-15 by breiler