winder/Universal-G-Code-Sender #2481: No visualizetion Windows 10

islonina, 2024-03-10:
### Version

2.1.6

### Hardware / Firmware

GRBL 1.1

### What happened

No visualization in the window. Tested on main computer connected to all my 3d printers and cnc. On backup one only for vinyl cuter all works fine. Bpth use the same zip file from github same Windows 10 Pro system.  Files are in .nc extension but github cant see them. 
[prover_v2_m6x8.txt](https://github.com/winder/Universal-G-Code-Sender/files/14551426/prover_v2_m6x8.txt)
Files are in .nc extension but github cant see them. 

### How to reproduce

Open file on UGS
No vizualization. Only for split cesound i can see it then black build plate

### Operating System

Windows 10 Pro 19045.3570 22H2

breiler, 2024-03-10:
Can you post the log file, you can find it it using the menu "Tools -> Open log directory".

breiler, 2024-03-11:
The log says that you are using version 2.1.5 which is installed at `C:\Program Files\Universal G-Code Sender`.

However throws a lot of errors where it can't find:
`C:\Users\islon\Downloads\win64-ugs-platform-app-2.1.5\ugsplatform-win\ide\modules\org-netbeans-modules-editor-fold-nbui.jar`

The settings might have become corrupt, you can try and delete this directory and restart UGS:
`C:\Users\islon\AppData\Roaming\ugsplatform`

islonina, 2024-03-11:
Somethong are bugged. I first use UGS on download folder then copy it on another folder. Now i redownlaod it againd and copy already to destination. All works fine now thx for the quide what i was missing before

closed 2024-03-14 by breiler