winder/Universal-G-Code-Sender #2483: When I'm using Setup Wizard the Maximum limit is set to a negative number

IVHead, 2024-03-12:
### Version

2.1.6

### Hardware / Firmware

GRBL 0.9

### What happened

I have been assigned a CNC Table using a gsheild v5b and Arduino UNO to repair and get functional. I have successfully gotten most aspects of the system functioning however some things on this software remain. I need to permanently disable hard limits but it keeps reenabling it, forcing me to go back to the "$$" menu and disabling the Boolean again. Next, when using the setup wizard the max distance I believe went into the negative, making it so I cant use the jog controller to send it in reverse but only forward. However the other functions such as G codes and homing can move them in reverse. There's just a lot I don't know. I would really appreciate some help. 

### How to reproduce

I really don't know how this happened, I am willing to send plenty more info and context if needed.

### Operating System

Windows 11

breiler, 2024-03-12:
Can you post the output from the ``$I` command?

IVHead, 2024-03-12:
"[0.9j.20160726:]" is what I got

IVHead, 2024-03-12:
also a picture
[image]

IVHead, 2024-03-12:
This is what my UI looks like.

breiler, 2024-03-12:
> I need to permanently disable hard limits but it keeps reenabling it, forcing me to go back to the "$$" menu and disabling the Boolean again.

Yes that is an old shortcoming to the setup wizard UI. Disabling the hard limits manually is currently the only work around.

> Next, when using the setup wizard the max distance I believe went into the negative, making it so I cant use the jog controller to send it in reverse but only forward.

Without your settings it is impossible to give any good answer. There are many parameters that are dependant of each other, such as the homing direction and this [compile time option](https://github.com/grbl/grbl/blob/master/grbl/config.h#L87).

In your screenshot it looks like you have jogged Y to +10 and it failed when you tried to jog Y back to 0. This is probably caused by a rounding error on the controller where it thinks that it exceeds the soft limit at position 0. Does it work if you lower the jog step size and jog Y closer to zero?

breiler, 2024-05-24:
Closing as we did not get further response from user

closed 2024-05-24 by breiler