winder/Universal-G-Code-Sender #2487: Continuous jogging runaway on X-Y when using joystick

lorenzo535, 2024-03-17:
### Version

2.1.5

### Hardware / Firmware

GRBL 1.1

### What happened

Jogging mode gets stuck when operating a joystick as an input device (xbox 360 gamepad), more precisely when input source is "analog jog" from analog joystick.

This happens following a prolonged jog command (i.e. keeping an analog joystick of the joypad deflected on a side for a while  to produce continuous jogging): when the joystick is released, often jog mode remains stuck (highlighted in green in GUI) , no buttons on GUI are responsive; joystick as well is no longer responsive. Only machine reset or reaching hard stop can stop the motion happening and make GUI responsive again: this shows that GRBL controller is not hanged, but simply UGS stays locked in perennial jog mode.

This can also be reproduced  if analog joystick is yanked from side to side (i.e. full left to full right deflection).

GRBL board used is ESP32 based with Grbl_Esp32 firmware.

Perhaps a simple fix would be to start a safety timer to kill jog every time a jog command is started: in this way either the user bringing back the joystick to center or the timer will ensure continuous jog is stopped, releasing UGS from its stuck in jog mode state.

### Operating System

ubuntu 22.04

### Anything else

Can be reproduced very often, especially if certain feed rates are selected: in my case for instance with a rate of 600 selected I can  almost every time reproduce it by yanking left right on the gamepad analog joystick.

breiler, 2024-03-17:
What is your configured maximum feed rate for each axis? (output from the command `$+`}?

What feed rate have you configured in the jog controller?
[image]

lorenzo535, 2024-03-18:
Runaway or jog mode lock is more likely to happen at feed rate 600. 
Here are my current grbl settings:
$0 = 10    (Step pulse time, microseconds)
$1 = 255    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 4    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 3    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 300.000    (Homing locate feed rate, mm/min)
$25 = 1000.000    (Homing search seek rate, mm/min)
$26 = 250.000    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$28 = 1920.000   
$30 = 10000.000    (Maximum spindle speed, RPM)
$31 = 0.000    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$38 = 1   
$40 = 1   
$N1=
$N0=
$34 = 0.000   
$35 = 0.000   
$36 = 1000.000   
$100 = 169.950    (X-axis travel resolution, step/mm)
$101 = 169.950    (Y-axis travel resolution, step/mm)
$102 = 318.191    (Z-axis travel resolution, step/mm)
$103 = 2000.000   
$104 = 2000.000   
$105 = 2000.000   
$110 = 6000.000    (X-axis maximum rate, mm/min)
$111 = 6000.000    (Y-axis maximum rate, mm/min)
$112 = 6000.000    (Z-axis maximum rate, mm/min)
$113 = 1000.000   
$114 = 1000.000   
$115 = 1000.000   
$120 = 100.000    (X-axis acceleration, mm/sec^2)
$121 = 100.000    (Y-axis acceleration, mm/sec^2)
$122 = 500.000    (Z-axis acceleration, mm/sec^2)
$123 = 200.000   
$124 = 200.000   
$125 = 200.000   
$130 = 2850.000    (X-axis maximum travel, millimeters)
$131 = 2720.000    (Y-axis maximum travel, millimeters)
$132 = 800.000    (Z-axis maximum travel, millimeters)
$133 = 300.000   
$134 = 300.000   
$135 = 300.000   
$140 = 0.250   
$141 = 0.250   
$142 = 0.250   
$143 = 0.250   
$144 = 0.250   
$145 = 0.250   
$150 = 0.125   
$151 = 0.125   
$152 = 0.125   
$153 = 0.125   
$154 = 0.125   
$155 = 0.125   
$160 = 16   
$161 = 16   
$162 = 16   
$163 = 16   
$164 = 16   
$165 = 16   
$170 = 16   
$171 = 16   
$172 = 16   
$173 = 16   
$174 = 16   
$175 = 16

breiler, 2024-03-18:
Ok, since your jog feed rate is lower than the maximum rate the this is not the problem that I was suspecting. I can't reproduce this.

You could try and update to UGS 2.1.6 and see if that helps. The gamepad library got updated in that version.

breiler, 2024-03-26:
What is your computer specification?

`cat /proc/cpuinfo`
`free -h`

lorenzo535, 2024-03-27:
heres the cpu info:
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 61
model name	: Intel(R) Core(TM) i5-5300U CPU @ 2.30GHz
stepping	: 4
microcode	: 0x2f
cpu MHz		: 500.000
cache size	: 3072 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 2
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 20
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm rdseed adx smap intel_pt xsaveopt dtherm ida arat pln pts vnmi md_clear flush_l1d
vmx flags	: vnmi preemption_timer invvpid ept_x_only ept_ad ept_1gb flexpriority tsc_offset vtpr mtf vapic ept vpid unrestricted_guest ple shadow_vmcs
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs taa itlb_multihit srbds mmio_unknown
bogomips	: 4589.38
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 61
model name	: Intel(R) Core(TM) i5-5300U CPU @ 2.30GHz
stepping	: 4
microcode	: 0x2f
cpu MHz		: 798.320
cache size	: 3072 KB
physical id	: 0
siblings	: 4
core id		: 1
cpu cores	: 2
apicid		: 2
initial apicid	: 2
fpu		: yes
fpu_exception	: yes
cpuid level	: 20
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm rdseed adx smap intel_pt xsaveopt dtherm ida arat pln pts vnmi md_clear flush_l1d
vmx flags	: vnmi preemption_timer invvpid ept_x_only ept_ad ept_1gb flexpriority tsc_offset vtpr mtf vapic ept vpid unrestricted_guest ple shadow_vmcs
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs taa itlb_multihit srbds mmio_unknown
bogomips	: 4589.38
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 2
vendor_id	: GenuineIntel
cpu family	: 6
model		: 61
model name	: Intel(R) Core(TM) i5-5300U CPU @ 2.30GHz
stepping	: 4
microcode	: 0x2f
cpu MHz		: 798.194
cache size	: 3072 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 2
apicid		: 1
initial apicid	: 1
fpu		: yes
fpu_exception	: yes
cpuid level	: 20
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm rdseed adx smap intel_pt xsaveopt dtherm ida arat pln pts vnmi md_clear flush_l1d
vmx flags	: vnmi preemption_timer invvpid ept_x_only ept_ad ept_1gb flexpriority tsc_offset vtpr mtf vapic ept vpid unrestricted_guest ple shadow_vmcs
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs taa itlb_multihit srbds mmio_unknown
bogomips	: 4589.38
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 3
vendor_id	: GenuineIntel
cpu family	: 6
model		: 61
model name	: Intel(R) Core(TM) i5-5300U CPU @ 2.30GHz
stepping	: 4
microcode	: 0x2f
cpu MHz		: 798.156
cache size	: 3072 KB
physical id	: 0
siblings	: 4
core id		: 1
cpu cores	: 2
apicid		: 3
initial apicid	: 3
fpu		: yes
fpu_exception	: yes
cpuid level	: 20
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm rdseed adx smap intel_pt xsaveopt dtherm ida arat pln pts vnmi md_clear flush_l1d
vmx flags	: vnmi preemption_timer invvpid ept_x_only ept_ad ept_1gb flexpriority tsc_offset vtpr mtf vapic ept vpid unrestricted_guest ple shadow_vmcs
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs taa itlb_multihit srbds mmio_unknown
bogomips	: 4589.38
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

and here's the free resources:
               total        used        free      shared  buff/cache   available
Mem:           7,6Gi       622Mi       5,9Gi       190Mi       1,1Gi       6,6Gi
Swap:          2,0Gi          0B       2,0Gi