winder/Universal-G-Code-Sender #2490: "Find" and "Replace" menu items grayed out

clydeshaffer, 2024-03-20:
### Version

2.1.5

### Hardware / Firmware

GRBL 1.1

### What happened

The "Find" and "Replace" options are grayed out under the "Edit" menu. I have a file loaded and wanted to search for certain strings but these appear to be disabled for some reason. If it's not because of a bug, it's at least unintuitive.

[image]

### How to reproduce

Opened a .nc file and opened the "Edit" drop down

### Operating System

Windows 11

breiler, 2024-03-21:
Thanks for the report, I have tried activating those features previously without success. I guess that I will need to deep dive into Netbeans API:s again to figure this one out.

closed 2024-03-21 by breiler

breiler, 2024-03-21:
Available in the latest nightly build: https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

PR #2491 (2024-03-21): Added library for editor search
Fixes #2490