winder/Universal-G-Code-Sender #2493: Fluidnc websocket connection failed.

John93-Blynk, 2024-03-22:
### Version

2.1.6

### Hardware / Firmware

FluidNC

### What happened

Hey there, I can't get Fluidnc to work using websocket. I'm using the latest version of Fluidnc (3.7.15), the latest version of the UGS platform (2.1.6), and when I hit connect, a message says ("cannot invoke "javax.websocket.session.close()" because "this.user.session" is null) appears. 
[image]
Am I missing something? Can you please tell me what should I do?

Greetings.

### Operating System

Windows 10

breiler, 2024-03-23:
Use the TCP connection driver instead, using the telnet protocol:
http://wiki.fluidnc.com/en/features/wifi_bt#telnet

The FluidNC websocket protocol is undocumented and does not work with UGS. For instance it assumes that the client sends pings to keep the connection alive.

John93-Blynk, 2024-03-23:
That worked like a charm. Thank you so much for your help.
Have a great day.

closed 2024-03-23 by John93-Blynk