winder/Universal-G-Code-Sender #2495: Allow soft limits without limit switches

agowa, 2024-03-26:
### Description

Hi, please allow specifying soft limits without limit switches. I also do not quite understand what good "soft limits" are when you've actual limit switches. For me they're only ever meaningful when you don't have limit switches. What am I missing? Can someone please explain?

breiler, 2024-03-26:
Read the documentation about soft limits here:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#20---soft-limits-boolean

Soft limits requires the machine to be homed, and for that you need limit switches. 

Additionally the limit switches can also work as hard limits, the setup wizard is currently depending on having hard limits enabled which is wrong. I have just not got around to fixing this yet. 

The purpose of being able to disable hard limits can be as a work around if your limit switches are being falsely triggered due to EMI from spindles or other high powered periphials.

agowa, 2024-03-29:
I don't know how else the thing I want is called. But all I want is to have the software be aware of the limits of the machine and not tell it to move beyond its capabilities.

I can see the status values in the software, so I should also be able to have the software bail out when I try to move to an offset that is outside of the safety region.

breiler, 2024-03-29:
It sounds like the feature you want is "soft limits" but without using a reliable method of referencing the machine zero position using limit switches.

It should technically be possible to make UGS keep track of machine position and limit sending any command that exceeds a defined max/min. This would require you to start your machine in a specific corner which will become your machine zero position. You would then need to configure UGS which directions it can move and how far.

This would however be a huge task to implement since it would require UGS to convert the command to be sent to a machine coordinate applying any offset (such as work or tool offset). 

It would be a handy feature for beginners getting started with CNC:s. But if I am being honest this is nothing I will prioritize since the time for me to implement this will far exceed the time for you to slap on three limit switches to your machine. 😄

agowa, 2024-03-29:
What do you mean without "a reliable method"? If manually zeroing before starting a job is reliable enough to do the job then it is also reliable enough for a soft limit. Or not?

breiler, 2024-03-30:
What do you mean by "If manually zeroing before starting a job"?

If you power on your controller when the spindle is positioned in your machine zero position that could technically be the same as a freshly homed machine and could be considered "reliable enough". But as stated in the GRBL documentation that I sent you earlier, you should use homing in order to use soft limits.

agowa, 2024-04-01:
With "If manually zeroing before starting a job" I mean ensuring to manually home the machine by using the Jog controller and a ruler to move it to where 0,0 should be (as almost always you want to engrave at a specific location you'll have to ensure that you start at the correct point either way).

* If the file is in Z- direction I also set Z=0 to be the top of the material otherwise (where Z0 is not the top but the bottom) I home it before putting the material. As the drill is often not "long enough" to reach all the way down this also often changes as I need to put a riser below it and the top of the riser would be Z=0 then.
* Depending on where the file expects the material to be I adjust X=0 and Y=0 to either be the center or one of the corners similarly.

breiler, 2024-04-02:
Ok I understand, you were writing about setting/zeroing work coordinates and not homing.

Lets say that you zero the work position somewhere on your aluminum block, how would UGS know how far it can move based on that position?

agowa, 2024-04-02:
Entering manually?

breiler, 2024-04-03:
I will not prioritize working on such a feature as it is already present in your hardware. I'll leave this feature request open in case anyone else wants to take a stab at it.

As I have already explained (and also linked to documentation), the only thing you need is three limit switches, enable homing so that the machine can find its zero position and enable soft limits defining the max work area.

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#22---homing-cycle-boolean
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#20---soft-limits-boolean

agowa, 2024-04-09:
> I will not prioritize working on such a feature

Understandable, as probably most people have the switches and sensors I assume?

> the only thing you need is three limit switches

The machine doesn't have them. There are also no mounting holes for them. However I ordered them, lets see if I can diy them on somewhere (and I also checked, you're right, the controller does have everything for them already. Even the connector is already soldered on).

missinglink, 2024-12-28:
Hello, I am using UGS with a CoreXY **Pen plotter**, this device doesn't have homing capabilities or switches, so I'm manually moving the pen to a far corner to set `0,0` with `G92 X0 Y0`.

I am then able to establish the bounds using the Jog Controller, when exceeding the bounds the device makes a jittery noise as it tries to move past its bounds.

The bounds for my device are 210x278.

When I enter these for the max travel distance (x-axis, y-axis) it seems to have no effect, it seems the soft-limits are not being honoured.

Is my issue related or am I doing something stupid?

[image]

breiler, 2024-12-29:
@missinglink you can't use soft limits without homing on GRBL:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#20---soft-limits-boolean

closed 2025-05-12 by breiler