winder/Universal-G-Code-Sender #2498: UGS Inverted Engraving

Strummerone, 2024-03-29:
### Version

2.1.6

### Hardware / Firmware

Grbl_ESP32

### What happened

Hi all.
Hopefully someone can help.
This is my first job...I'm using a home made but basic XYYZ configuration. (Router)
I'm trying to engrave some text onto some plywood (Test piece). Absolutely delighted with the results except for the fact that it reads inverted! It looks fantastic if viewed in the mirror.... Which is not the desired effect. Correct way round would be nice.

SET UP: Fusion 360 > design > some text > Manufacture> create Gcode. No issues.
Hardware / Firmware: Mks dlc32 running Fluidnc via wifi.
UGS > Setup > Wizard > Checked motor direction using the wizard >all ok.
UGS > Visualizer display reads correct and tracks the tool in the correct direction..
UGS > Tried reverse motor direction in setup, made no difference.
Any ideas would be great..

### Operating System

MacBook Pro M3

breiler, 2024-03-30:
Can you post the output from the console when you enter these commands `$I` and `$$`

I think that you were on the right track inverting the stepper motor direction. You did not specify what kind of machine you built. 

But a common mistake users make when they build these types of machines where the gantry is stationary is that the Y-axis should be inverted:
[image]

As opposed to these kinds of machines where the material is stationary and both the XYZ axes are moving:
[image]

Strummerone, 2024-03-30:
Hi Breiler,  Thanks for the reply.
The machine has stationary material, all axis move.
I couldn't wait to see if the machine worked yesterday as I've been working on it for a year. I'm in the process of wiring in limit switches at the moment but can still access the firmware. It would be good to get it working properly once I've done that.
Here is the info you requested.
[$ info.txt](https://github.com/winder/Universal-G-Code-Sender/files/14812107/info.txt)

Strummerone, 2024-03-31:
Hi Breiler, 

I managed to sort the problem. For the record, I swapped the coil wires on the Y axis motors which did the trick.
Regards all.

closed 2024-04-02 by breiler