winder/Universal-G-Code-Sender #2499: Designer objects change incorrectly

dmilton, 2024-03-29:
### Version

2.1.6

### Hardware / Firmware

GRBL 1.1

### What happened

Take any design with more than one design object. Try to change the start depth, target depth, or cut type and it won't stick.

### How to reproduce

Start a new document. Create a rectangle object. Create a second object. In the Design objects, select the first object and change the cut type, start depth and/or target depth. Select the second object. Both now have the same attributes. Go back to the first object and try to change it. You essentially cannot. As soon as you select a different object it takes on the properties of the other.

### Operating System

MacOS Monterey 12.7.4, Ventura 3.6.6, and Sonoma 14.4.1.

### Anything else

This problem occurs all the time. At the moment I haven't figured out how to get around it. Every once in a while changes to a design object will stick but I can't tell if that's just the previously selected object taking on the properties of the newly selected object or not.

dmilton, 2024-03-29:
Just tested with 2.1.5 and the problem is similar there. This time the cut type stays the same but the start depth and target depth keeps changing to the previously selected object. Just click between two objects is all it takes.

garasimeria, 2024-03-30:
on my 4 computers with Windows 10, it’s the same problem

breiler, 2024-05-24:
I think that I solved this in #2525. It is available in the latest nightly build: 
https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

garasimeria, 2024-05-25:
Unfortunately version 2.0 snapshot/2024-05-24 did not solve the problem
either

În vin., 24 mai 2024 la 08:27, Joacim Breiler ***@***.***> a
scris:

breiler, 2024-05-25:
Darn it, I managed to reproduce this now too. But it doesn't happen always.

garasimeria, 2024-05-27:
Indeed, it no longer modifies Cut Type but changes target depth

When I right click then Select all, the 5 dots disappear for x,y coordinates

În sâm., 25 mai 2024 la 10:43, Joacim Breiler ***@***.***> a
scris:

closed 2025-02-15 by breiler