winder/Universal-G-Code-Sender #2501: Visualizer not functioning in relative mode

ehouais, 2024-04-03:
### Version

2.1.6

### Hardware / Firmware

GRBL 1.1

### What happened

When plotting a GCode program in relative mode (G91), nothing is shown in the visualizer, though the program is valid and successfully plotted on my iDraw A3. When I switch to absolute coordinates using the code G90, the visualizer shows the path as it should.

### How to reproduce

Simple example program in relative mode that is not visalized properly:
```
G92 X0 Y0
G90
G1 F10000 Z0.5
G91 
G1 X50 Y-50
G1 F5000 Z4.5
G1 F2000
G1 X-10 Y0
G1 X0 Y10
G1 X10 Y0
G1 X0 Y-10
G1 F10000 Z-4.5
G1 F8000
G1 X-50 Y50
```

Equivalent program in absolute mode that is visualized properly
```
G92 X0 Y0
G90
G1 F10000 Z0.5
G1 X50 Y-50
G1 F5000 Z4.5
G1 F2000
G1 X40 Y-50
G1 X40 Y-40
G1 X50 Y-40
G1 X50 Y-50
G1 F10000 Z-4.5
G1 F8000
G1 X0 Y0
```

### Operating System

Mac mini M2 with macOS Sonoma

breiler, 2024-04-03:
How did you generate this program?

ehouais, 2024-04-03:
Home-made generation pipeline. I find relative moves more suitable to my algorithmic experiments.

ehouais, 2024-04-03:
OK, tested it and it works now. Thanks for your quick response !