winder/Universal-G-Code-Sender #2503: UGS cannot control the spindle

chengeV587, 2024-04-03:
### Version

2.1.6

### Hardware / Firmware

FluidNC

### What happened

Sending M3 S5000, M5 commands through the console and switching the Spindle button of the overrides panel failed to control the spindle，an attempt has been made to switch to tool number 0 by T0。Jog control works fine。
I've tried other Gcode senders like Openbuilds, CNCjs, and they all work fine, but UGS is my favorite。
[image]

### Operating System

Window Version: Windows 11

### Anything else

When I connect to UGS and query the state of the machine with the $$instruction, I get this result：
[image]
Where $30=255 is incorrect and $30=12000 is correct。
After connecting UGS, other Gcode Sender such as CNCjs will get the same result of $$ instruction query, and the spindle can not be controlled. The machine must be restarted.
The result of the $$ query sent by CNCjs is as follows:
[image]

breiler, 2024-04-04:
This worked just fine for me, recheck your settings

My version
```
> $I
[VER:3.7 FluidNC v3.7.16:]
[OPT:PHS]
[MSG:Machine: New configuration]
ok
```

My settings
```
> $$
$22=1
$21=0
$20=0
$102=800.000
$101=800.000
$100=800.000
$112=1200.000
$111=1200.000
$110=1200.000
$122=15.000
$121=15.000
$120=15.000
$132=1000.000
$131=1000.000
$130=1000.000
$10=1
$30=12000
ok
```

Spindle control works, however the M5 command for stopping the spindle will still make the status report a spindle speed. I will post a bug report to the FluidNC team.
```
> m3 S12000
ok
> ?
<Idle|MPos:0.000,0.000,0.000|FS:0,12000|Pn:Y>
ok
> m5
ok
> ?
<Idle|MPos:0.000,0.000,0.000|FS:0,12000|Pn:Y>
ok
> m3 S0
ok
> ?
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
ok
```

My config file
```yaml
> $config/dump
board: None
name: New configuration
meta: 
stepping: 
  engine: RMT
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 12

spi: 
  miso_pin: NO_PIN
  mosi_pin: NO_PIN
  sck_pin: NO_PIN

sdcard: 
  cs_pin: NO_PIN
  card_detect_pin: NO_PIN
  frequency_hz: 8000000

kinematics: 
  Cartesian: 

axes: 
  shared_stepper_disable_pin: gpio.13
  shared_stepper_reset_pin: NO_PIN
  x: 
    steps_per_mm: 800.000000
    max_rate_mm_per_min: 1200.000000
    acceleration_mm_per_sec2: 15.000000
    max_travel_mm: 1000.000000
    soft_limits: false
    homing: 
      cycle: 2
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000000
      feed_mm_per_min: 50.000000
      seek_mm_per_min: 200.000000
      settle_ms: 250
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.17:low:pu
      hard_limits: false
      pulloff_mm: 2.000000
      standard_stepper: 
        step_pin: gpio.12
        direction_pin: gpio.14:low
        disable_pin: NO_PIN

  y: 
    steps_per_mm: 800.000000
    max_rate_mm_per_min: 1200.000000
    acceleration_mm_per_sec2: 15.000000
    max_travel_mm: 1000.000000
    soft_limits: false
    homing: 
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000000
      feed_mm_per_min: 50.000000
      seek_mm_per_min: 200.000000
      settle_ms: 250
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.4:low:pu
      hard_limits: false
      pulloff_mm: 2.000000
      standard_stepper: 
        step_pin: gpio.26
        direction_pin: gpio.15
        disable_pin: NO_PIN

  z: 
    steps_per_mm: 800.000000
    max_rate_mm_per_min: 1200.000000
    acceleration_mm_per_sec2: 15.000000
    max_travel_mm: 1000.000000
    soft_limits: false
    homing: 
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000000
      feed_mm_per_min: 50.000000
      seek_mm_per_min: 200.000000
      settle_ms: 250
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.16:low:pu
      hard_limits: false
      pulloff_mm: 1.000000
      standard_stepper: 
        step_pin: gpio.27
        direction_pin: gpio.33:low
        disable_pin: NO_PIN

control: 
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN

coolant: 
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe: 
  pin: gpio.32:low:pu
  toolsetter_pin: NO_PIN
  check_mode_start: false

macros: 
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 
  after_homing: 
  after_reset: 
  after_unlock: 

start: 
  must_home: true
  deactivate_parking: false
  check_limits: false

parking: 
  enable: false
  axis: Z
  target_mpos_mm: -5.000000
  rate_mm_per_min: 800.000000
  pullout_distance_mm: 5.000000
  pullout_rate_mm_per_min: 250.000000

user_outputs: 
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN

PWM: 
  pwm_hz: 5000
  direction_pin: NO_PIN
  output_pin: gpio.2
  enable_pin: gpio.22
  disable_with_s0: true
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 100
  speed_map: 0=0.00% 12000=100.00%
  off_on_alarm: false

arc_tolerance_mm: 0.002000
junction_deviation_mm: 0.010000
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
ok
```

chengeV587, 2024-04-04:
Maybe my SpeedMap is too long. After I changed the SpeedMap to be the same as yours (10V as your PWM) , M3 and M5 can control the spindle normally.
[image]
However, the Spindle toggle button in the overrides panel is still unresponsive (no command is sent in the Console after clicking).
The sliding bar can control the spindle speed

breiler, 2024-04-04:
The spindle override feature implementation in GRBL is unintuitive and I was actually considering to remove it from the overrides panel. 

From the GRBL protocol specification you can read the following:
```
0x9E : Toggle Spindle Stop

Toggles spindle enable or disable state immediately, but only while in the HOLD state.
The command is otherwise ignored, especially while in motion. This prevents accidental disabling during a job that can either destroy the part/machine or cause personal injury. Industrial machines handle the spindle stop override similarly.
When motion restarts via cycle start, the last spindle state will be restored and wait 4.0 seconds (configurable) before resuming the tool path. This ensures the user doesn't forget to turn it back on.
While disabled, spindle speed override values may still be altered and will be in effect once the spindle is re-enabled.
If a safety door is opened, the DOOR state will supersede the spindle stop override, where it will manage the spindle re-energizing itself upon closing the door and resuming. The prior spindle stop override state is cleared and reset.
```

closed 2024-04-04 by chengeV587