winder/Universal-G-Code-Sender #2506: Cannot find 'Platform*' folder!

brqzz, 2024-04-15:
### Version

2.1.6

### Hardware / Firmware

Grbl-Mega

### What happened

In short, I was using it for the past week, and it worked fine, now that I have to document it for my bachelor it decided to stop working, tried all the steps I could find in previous issues, and also I'm not trying to run it directly from .zip 

I've tried 2.1.6, 2.1.5 and Lite versions. I have also tried removing my old configuration files, but none was created at launch.

I'm adding my log files, but I have to mention, that their log is back from 04-11 (message.1) and 04-12 (message.2) and if i remember correctly it still worked back then

[messages (2).log](https://github.com/winder/Universal-G-Code-Sender/files/14977378/messages.2.log)
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/14977379/messages.log)

### Operating System

I'm using is Windows 11

breiler, 2024-04-15:
Do you get any error messages when you start the program. For instance you wrote `Cannot find 'Platform*' folder` in the title, what do you mean by that?

Make sure the location of UGS is placed in a directory path without any special characters (such as `åäö{[]}!?;:^~` and so on).

breiler, 2024-04-16:
According to the log the application was in the directory `C:\Users\Administrator\Desktop\8 kursas\Bakalauras\win64-ugs-platform-app-2.1.6\` last week.

Nice that you got it working again, I am closing this.

closed 2024-04-16 by breiler