winder/Universal-G-Code-Sender #2507: gcode file it open, appear 1/4 of second and disappear.

EnriqueTornel, 2024-04-18:
### Version

2.1.2

### Hardware / Firmware

GRBL 0.9

### What happened

When I open a gcode file it appears brevely and disappear. It was working fine untill yesterday with the same gcode file.

### Operating System

windows 10

breiler, 2024-04-18:
Can you post a screen shot and the log file from `C:/Users/[your username>]/AppData/Roaming/ugsplatform/var/log/messages.log`

Vegsaad, 2024-04-20:
I am having the exact same problem on my laptop with the latest V2.1.6 download but on my main PC no problem. using the same gcode grbl files on both. A much earlier version of UGS works fine on all my machines.

Vegsaad, 2024-04-20:
I have just fixed my problem. It appears to have been due to file corruption of the setting files. I had inadvertently left the old version running after installing the lates version. Your help on another discussion lead me to the answer: delete the setting files as per your advice here: 

https://github.com/winder/Universal-G-Code-Sender/discussions/2476#discussioncomment-8681125

[image]

breiler, 2024-04-23:
Closing as we didn't receive any response from original poster. 

Thanks for the feedback @Vegsaad

closed 2024-04-23 by breiler