winder/Universal-G-Code-Sender #2513: Universal g-code sender doesn't want to open .gcode files and crashes when saving clipart

wmoscon1, 2024-05-06:
### Version

2.1.6

### Hardware / Firmware

GRBL 1.1

### What happened

Hello.. when trying to open .gcode or .ngc files, the program cannot load the files and closes the tab and does not load, in relation to the clipart, the moment I go to save it, it crashes the entire system, thanks.
Note: sorry for my English, I'm Brazilian

### How to reproduce

[image]

### Operating System

Windowns 10

breiler, 2024-05-07:
Can you share your log file? Go to the menu "Tools > Open log directory" and attach the ?` `messages.log` here.

wmoscon1, 2024-05-07:
he follows..

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/15228546/messages.log)

wmoscon1, 2024-05-07:
If you want me to delete it and generate it again, just let me know.

Another thing... by chance it doesn't generate cash, maybe it would just be me cleaning it, because it worked on the other machine, but on the one I'm on it doesn't open. grateful

wmoscon1, 2024-05-07:
[image]

wmoscon1, 2024-05-08:
I managed to solve the problem, however, on the two machines I installed, the gcode clipart is not generated correctly.
grateful!!

breiler, 2024-05-08:
What do you mean that the gcode clipart is not generated? Can you post a screenshot with the error?

wmoscon1, 2024-05-08:
When I create clipart, for example, I place an ant, when I send it to the machine it asks to save what was done and then generates a gcode without positions.

[image]

[image]

Or of course, it could be me who doesn't have much experience with the application yet.

grateful

breiler, 2024-05-08:
Ah, select the ant shape and set the cut type and the target depth. Then press save and you should see it in the 3D visualiser.

wmoscon1, 2024-05-08:
Thank you very much, it worked!

wmoscon1, 2024-05-08:
One more question, how do I increase the memory consumed by the application?
Why does the app crash a lot when manipulating it!

breiler, 2024-05-08:
Can I have a fresh log file?

wmoscon1, 2024-05-08:
Of course, follow..

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/15253456/messages.log)

breiler, 2024-05-08:
I don't think that memory is an issue, from the log I can see that it will use up to 8Gb which is more than enough to be able to run UGS. Next time it crashes, post the latest log file from the directory `C:\Users\wmosc\AppData\Roaming\ugsplatform\var\log\`

wmoscon1, 2024-05-08:
I had to close it now through the task manager because it didn't respond, here's the most recent generated file...

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/15253580/messages.log)

wmoscon1, 2024-05-08:
The moment I resize the image it stops responding

wmoscon1, 2024-05-08:
Here it got stuck right when the image was being resized, it was shrinking the image.

[image]

wmoscon1, 2024-05-08:
The application stops responding.

[image]

breiler, 2024-05-08:
Did you import a DXF or SVG file? May I have the source? It looks like it contains a lot of points which could explain this problem.

wmoscon1, 2024-05-08:
I imported a png, an image, this would be the one below.

[image]

wmoscon1, 2024-05-08:
the site won't let me load the file that ugs saves.

wmoscon1, 2024-05-08:
[image]

[image]

breiler, 2024-05-09:
I understand, the settings you were using will create very many small tiny objects which UGS will have problems applying the scaling operation to. 

I think that if you wanted to carve this it will take a loooong time. So the key is to simplify. 

* Generally you want as few layers as possible, otherwise you will cut the same shape multiple times.
* Color quantization is how the colors will be grouped together.
* Smothing of lines, curves and filter noise will remove small imperfections in the image (such as compression artefacts on JPGS)
* Blur will further simplify the vector result.

With these settings I get somewhat more usable vector graphics:
[image]
[image]

wmoscon1, 2024-05-09:
Perfect... thank you very much for your help.

wmoscon1, 2024-05-09:
[image]

wmoscon1, 2024-05-09:
For some reason it was unable to generate the design when passed to the machine, it was lost.

[image]

wmoscon1, 2024-05-09:
[image]

breiler, 2024-05-12:
I am closing this as it seems that you are up and running.

closed 2024-05-12 by breiler