winder/Universal-G-Code-Sender #2514: Feedrate not applied correctly after gcode program rotation in UGS

lorenzo535, 2024-05-08:
### Version

2.1.6

### Hardware / Firmware

GRBL 1.1

### What happened

When a gcode program is rotated left or right within UGS, upon its execution the feedrates are not applied correctly anymore:

For instance:
if not rotated, i.e. using the gcode file as loaded 
N80G1Z-3F250
N90G2X116.2635Y83.7365I16.2635J-16.2635F2000
the G2 command will be executed correctly at F2000

If the gcode program is rotated (see here copy of console output):
>>> F0.0S20000.0G0X155.4961Y16.8393Z5
ok
ok
>>> F250.0S20000.0G1X155.4961Y16.8393Z-3
ok
>>> N90F2000
ok
[MSG:Tool No: 1]
ok
>>> F250.0S20000.0G1X155.4256Y16.9101Z-3
ok
>>> F250.0S20000.0G1X155.3554Y16.9813Z-3
ok
>>> F250.0S20000.0G1X155.2855Y17.0527Z-3
ok
ok
>>> F250.0S20000.0G1X155.2159Y17.1244Z-3

The feedback after command at line N90 shows that the feed rae applied is 250, i.e. the feed rate of line N80

### How to reproduce

Run the gcode program below:  notice that the Z plunge operations (like N80, N110, ....) are indeed done at F250, and the G1 G2 commands are indeed executed at F2000.

Then rotate the program in UGS and run it again: command N90 will not be executed at F2000 but at the feedrate of the prior command (F250).

N10 G21 G17 G90 G40 G49 G80
N20 G91.1
N30 G0 Z20
N40 G0 X83.7365 Y116.2635
N50 T1
N60 S20000 M03
N70 G0 Z5
N80 G1 Z-3 F250
N90 G2 X116.2635 Y83.7365 I16.2635 J-16.2635 F2000
N100 G2 X83.7365 Y116.2635 I-16.2635 J16.2635
N110 G1 Z-6 F250
N120 G2 X116.2635 Y83.7365 I16.2635 J-16.2635 F2000
N130 G2 X83.7365 Y116.2635 I-16.2635 J16.2635
N140 G1 Z5
N150 G0 X215.6565 Y104.8622
N160 G1 Z-4.5 F250
N170 G2 X261.6565 I23 J0 F2000
N180 G2 X215.6565 I-23 J0
N190 G1 Z-9 F250
N200 G2 X232.9004 Y127.1303 I23 J0 F2000
N210 G1 Z-5
N220 G2 X244.7644 Y127.0363 I5.7561 J-22.2681
N230 G1 Z-9 F250
N240 G2 X261.6565 Y104.8622 I-6.1079 J-22.1742 F2000
N250 G2 X244.4786 Y82.6113 I-23 J0
N260 G1 Z-5
N270 G2 X232.6144 Y82.67 I-5.8221 J22.2509
N280 G1 Z-9 F250
N290 G2 X215.6565 Y104.8622 I6.0421 J22.1922 F2000
N300 G1 Z5
N310 G0 Z20
N320 M05
N330 M09
N340 M30

### Operating System

ubuntu 22.04

### Anything else

Every time

breiler, 2024-05-08:
Thanks for the report.

As a workaround you can place the feed rate on an empty line before:
```gcode
N80 G1 Z-3 F250
N85 F2000
N90 G2 X116.2635 Y83.7365 I16.2635 J-16.2635 F2000
```

lorenzo535, 2024-05-08:
Yes indeed , however the gcode is generated by a third party software that doesn't allow to specify to add that systematically. Just for my own curiosity, did you manage to reproduce the behaviour?

breiler, 2024-05-08:
Yes, with your example gcode this was easily repeatable, thanks! I will not have time to look into why this is happening until maybe this weekend, but I thought that it would be nice to have some work around in the meantime.

lorenzo535, 2024-05-08:
Sure indeed, and thank you for your support. By the way the rotate function is particularly useful.