winder/Universal-G-Code-Sender #2517: Program freezes when arc radii are too large (and python fix)

gatornuke235, 2024-05-10:
### Version

2.1.6

### Hardware / Firmware

GRBL 1.1

### What happened

When loading a gcode file with an intricate geometry  that may have extremely wide radius arcs the UGS sender freezes. older versions would send the router off in random directions when that line was executed.

As a result I wrote a python script to scrub the gcode file and change the G2 or G3 commands to a G1 line. After all, any short arc segments with very large radius are basically a straight line:

`import sys

def process_gcode(input_file, output_file, threshold):
    threshold = float(threshold)
    with open(input_file, 'r') as file:
        lines = file.readlines()

    with open(output_file, 'w') as file:

    
        for line in lines:
            if line.startswith(('G2', 'G3')):
                exceeded = False
                parts = line.split()
                
                for part in parts[1:]:
                    if part.startswith(('I', 'J', 'K')):
                        value = float(part[1:])
                        if abs(value) > threshold:
                            exceeded = True
                        
                if (exceeded):            
                    new_line = 'G1' + ' (' + parts[0] + ')' # Overrite the original G2 or G3 command
                    for part in parts[1:]:
                        if part.startswith(('X', 'Y', 'Z', 'F')):
                            new_line += ' ' + part
                        elif part.startswith(('I', 'J', 'K')):
                            new_line += ' (' + part + ')'  # Comment out arc
                    file.write(new_line + '\n')
                else:
                    file.write(line)
            else:
                file.write(line)

if __name__ == '__main__':
    args = sys.argv
    globals()[args[1]](*args[2:])`

This is callable from the command shell as such:
`python cleaner.py process_gcode input.gcode output.gcode 800000`

Which will change this:
`G3 X-35.429 Y-19.142 Z-2.000 I3486934.545 J205113.019 K0.000 F500.000`
into this:
`G1 (G3) X-35.429 Y-19.142 Z-2.000 (I3486934.545) (J205113.019) (K0.000) F500.000`

### How to reproduce

include this line in any gcode:
`G3 X-35.429 Y-19.142 Z-2.000 I3486934.545 J205113.019 K0.000 F500.000`

### Operating System

Windows 11

### Anything else

Every time

breiler, 2024-05-10:
Can you post a sample gcode program with the problem. This could be related to the visualizer which will take an arc and split it up into several small lines.

gatornuke235, 2024-05-10:
Sure. this should cause the error:
```
(Exported by FreeCAD)
(Post Processor: grbl_post)
(Output Time:2024-05-10 01:52:00.747480)
(Begin preamble)
G17 G90
G21
(Begin operation: Fixture)
(Path: Fixture)
G54
(Finish operation: Fixture)
(Begin operation: 4th-inch-endmill-6.35mm-001)
(Path: 4th-inch-endmill-6.35mm-001)
(4th-inch-endmill-6.35mm)
(Begin toolchange)
( M6 T1 )
M3 S20000
(Finish operation: 4th-inch-endmill-6.35mm-001)
(Begin operation: Pocket_Shape)
(Path: Pocket_Shape)
(Pocket_Shape)
G0 Z6
G0 X0 Y0
G0 X-18.550 Y52.769
G0 X-18.550 Y52.769 Z4
G1 X-18.550 Y52.769 Z-4 F200
G2 X-18.725 Y55.370 Z-4 I40.631 J4.042 K0 F700
G1 X-18.725 Y74.856 Z-4 F700
G3 X-24.456 Y74.256 Z-4 I40.691 J-416.520 K0 F700
G1 X-27.344 Y73.818 Z-4 F700
G2 X-27.306 Y72.929 Z-4 I-19.200 J-1.250 K0 F700
G2 X-27.306 Y52.107 Z-4 I-2426220.106 J-10.411 K0 F700
G2 X-21.382 Y51.786 Z-4 I2.804 J-3.066 K0 F700
G2 X-19.584 Y52.648 Z-4 I3.996 J-6.028 K0 F700
G2 X-18.550 Y52.769 Z-4 I0.974 J-3.837 K0 F700
G0 Z6
G0 X-27.306 Y54.252 Z6
G0 X-27.306 Y54.252 Z4
G1 X-27.306 Y54.252 Z-4 F200
G1 X-27.306 Y52.107 Z-4 F700
G2 X-25.974 Y52.919 Z-4 I2.688 J-2.915 K0 F700
G1 X-27.306 Y54.252 Z-4 F700
G1 X-27.306 Y60.089 Z-4 F700
G1 X-19.797 Y52.580 Z-4 F700
G2 X-18.550 Y52.769 Z-4 I1.106 J-3.076 K0 F700
G2 X-18.719 Y55.147 Z-4 I237.203 J18.072 K0 F700
G2 X-18.725 Y57.345 Z-4 I31.094 J1.183 K0 F700
G1 X-27.306 Y65.926 Z-4 F700
G3 X-27.306 Y71.763 Z-4 I-4458.754 J2.919 K0 F700
G1 X-18.725 Y63.182 Z-4 F700
G1 X-18.725 Y69.019 Z-4 F700
G1 X-24.028 Y74.321 Z-4 F700
G2 X-18.725 Y74.856 Z-4 I401.296 J-3952.942 K0 F700
G0 Z6
G0 Z6
G0 X-18.725 Y74.856
G0 X-51.416 Y68.830
G0 X-51.416 Y68.830 Z4
G1 X-51.416 Y68.830 Z-4 F200
G1 X-54.967 Y67.785 Z-4 F700
G1 X-57.676 Y66.988 Z-4 F700
G3 X-57.755 Y66.965 Z-4 I0.895 J-3.186 K0 F700
G3 X-58.363 Y66.770 Z-4 I2.430 J-8.631 K0 F700
G1 X-58.363 Y66.770 Z-4 F700
G2 X-59.959 Y66.238 Z-4 I-1695759.683 J5087278.619 K0 F700
G1 X-60.746 Y65.976 Z-4 F700
G2 X-60.535 Y62.463 Z-4 I-216.044 J-14.692 K0 F700
G2 X-60.534 Y62.268 Z-4 I-9.979 J-0.170 K0 F700
G1 X-60.534 Y46.894 Z-4 F700
[... 390 lines cut ...]
G2 X47.101 Y70.710 Z-4 I-0.380 J-1.981 K0 F700
G3 X48.058 Y70.460 Z-4 I2.621 J8.070 K0 F700
G1 X50.028 Y69.945 Z-4 F700
G1 X42.535 Y62.452 Z-4 F700
G1 X42.535 Y56.615 Z-4 F700
G1 X51.328 Y65.408 Z-4 F700
G1 X51.328 Y60.671 Z-4 F700
G3 X51.439 Y59.682 Z-4 I5.266 J0.092 K0 F700
G1 X42.535 Y50.778 Z-4 F700
G3 X42.594 Y49.550 Z-4 I18.680 J0.278 K0 F700
G2 X42.747 Y46.903 Z-4 I-28.146 J-2.954 K0 F700
G2 X44.583 Y46.988 Z-4 I1.097 J-3.814 K0 F700
G1 X53.225 Y55.631 Z-4 F700
G3 X53.490 Y55.429 Z-4 I1.401 J1.556 K0 F700
G3 X51.328 Y51.808 Z-4 I2.012 J-3.657 K0 F700
G3 X51.328 Y47.896 Z-4 I292.387 J-1.969 K0 F700
G0 Z6
G0 Z6
(Finish operation: Pocket_Shape)
(Begin postamble)
M5
G17 G90
M2

```

gatornuke235, 2024-05-10:
I should also mention that i have to omit the commented out sections in version 20240308 (2.1.6), 
```
java.lang.StringIndexOutOfBoundsException: String index out of range: 0
	at java.base/java.lang.StringLatin1.charAt(Unknown Source)
	at java.base/java.lang.String.charAt(Unknown Source)
	at com.willwinder.universalgcodesender.gcode.GcodePreprocessorUtils.extractWord(GcodePreprocessorUtils.java:443)
	at com.willwinder.universalgcodesender.gcode.GcodePreprocessorUtils.parseCoord(GcodePreprocessorUtils.java:455)
	at com.willwinder.universalgcodesender.gcode.GcodePreprocessorUtils.updatePointWithCommand(GcodePreprocessorUtils.java:206)
	at com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils.handleGCode(GcodeParserUtils.java:263)
	at com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils.processCommand(GcodeParserUtils.java:152)
	at com.willwinder.universalgcodesender.gcode.GcodeParser.addCommand(GcodeParser.java:147)
	at com.willwinder.universalgcodesender.gcode.GcodeParser.addCommand(GcodeParser.java:134)
	at com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils.preprocessAndWrite(GcodeParserUtils.java:400)
	at com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils.processAndExportText(GcodeParserUtils.java:442)
	at com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils.processAndExport(GcodeParserUtils.java:381)
	at com.willwinder.universalgcodesender.model.GUIBackend.preprocessAndExportToFile(GUIBackend.java:152)
	at com.willwinder.universalgcodesender.model.GUIBackend.initializeProcessedLines(GUIBackend.java:785)
	at com.willwinder.universalgcodesender.model.GUIBackend.processGcodeFile(GUIBackend.java:380)
	at com.willwinder.universalgcodesender.model.GUIBackend.setGcodeFile(GUIBackend.java:471)
[catch] at com.willwinder.ugs.nbp.editor.EditorUtils.openFile(EditorUtils.java:45)
	at com.willwinder.ugs.nbp.editor.SourceMultiviewElement.componentOpened(SourceMultiviewElement.java:72)
	at org.netbeans.core.multiview.MultiViewPeer.showCurrentElement(MultiViewPeer.java:471)
	at org.netbeans.core.multiview.MultiViewPeer.peerComponentOpened(MultiViewPeer.java:357)
	at org.netbeans.core.multiview.MultiViewCloneableTopComponent.componentOpened(MultiViewCloneableTopComponent.java:129)
	at org.openide.windows.WindowManager.componentOpenNotify(WindowManager.java:282)
	at org.netbeans.core.windows.WindowManagerImpl.notifyTopComponentOpened(WindowManagerImpl.java:1192)
	at org.netbeans.core.windows.Central.addModeOpenedTopComponent(Central.java:773)
	at org.netbeans.core.windows.ModeImpl.addOpenedTopComponent(ModeImpl.java:336)
	at org.netbeans.core.windows.WindowManagerImpl.topComponentOpenAtTabPosition(WindowManagerImpl.java:1305)
	at org.netbeans.core.windows.WindowManagerImpl.topComponentOpen(WindowManagerImpl.java:1262)
	at org.openide.windows.TopComponent.open(TopComponent.java:473)
	at org.openide.windows.TopComponent.open(TopComponent.java:453)
	at org.openide.windows.CloneableOpenSupport.openCloneableTopComponent(CloneableOpenSupport.java:183)
	at org.openide.windows.CloneableOpenSupport$1.run(CloneableOpenSupport.java:82)
	at org.netbeans.modules.openide.util.NbMutexEventProvider$Event.doEvent(NbMutexEventProvider.java:97)
	at org.netbeans.modules.openide.util.NbMutexEventProvider$Event.writeAccess(NbMutexEventProvider.java:62)
	at org.netbeans.modules.openide.util.LazyMutexImplementation.writeAccess(LazyMutexImplementation.java:56)
	at org.openide.util.Mutex.writeAccess(Mutex.java:292)
	at org.openide.windows.CloneableOpenSupport.open(CloneableOpenSupport.java:79)
	at org.openide.text.CloneableEditorSupport.open(CloneableEditorSupport.java:412)
	at com.willwinder.ugs.nbp.core.actions.OpenFileAction.actionPerformed(OpenFileAction.java:61)
	at com.willwinder.ugs.nbp.core.actions.OpenAction.openFile(OpenAction.java:92)
	at java.base/java.util.Optional.ifPresent(Unknown Source)
	at com.willwinder.ugs.nbp.core.actions.OpenAction.actionPerformed(OpenAction.java:87)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
	at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
	at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
	at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
	at java.desktop/javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
[... cut ...]
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(Unknown Source)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

```

breiler, 2024-05-10:
I had no problems creating a gcode file with the contents and running it. Can you attach the actual gcode file, I am suspecting a UTF-8 BOM character problem.

gatornuke235, 2024-05-11:
After updating java and assigning more memory to UGS (-J-Xms1024m) I'm no longer having the crashes, now it just behaves like the older version, sending the router in a straight line to infinity unless i hit the stop button before it reaches the limit when a bad instruction is called.

The last lines shown in the console before I stopped the excursion are:
```
>>> G3X-57.755Y66.965Z-4I0.895J-3.186K0F4000
ok
ok
>>> G3X-58.363Y66.770Z-4I2.430J-8.631K0F4000
ok
>>> G1X-58.363Y66.770Z-4F4000
ok
>>> G2X-59.959Y66.238Z-4I-1695759.683J5087278.619K0F4000
ok
>>> G1X-60.746Y65.976Z-4F4000
ok
>>> G2X-60.535Y62.463Z-4I-216.044J-14.692K0F4000
ok

**** Canceling file transfer. ****

**** Pausing file transfer. ****

ok
Grbl 1.1h for MillRight CNC Carve King 2 ['$' for help] 
```

Note the very large I and J values, which is why I got suspicious of these in the first place: I-1695759.683 J5087278.619. I suspected some kind of data overflow issue.

Executing the modified file runs the job to completion with no excursions. See files attached.

[FnI_2_6mm-test-files.zip](https://github.com/winder/Universal-G-Code-Sender/files/15281103/FnI_2_6mm-test-files.zip)

breiler, 2024-05-11:
Ok so there are two problems. I was completely focusing on the loading/freezing problem which does still not happen for me with the default settings in UGS (2.1.5 or 2.1.6). 

The setting you've changed is setting the minimum required memory, not increasing the maximum. By default Java will be able to use maximum 1/4 of the total memory. So if you have < 4GB of ram this setting could potentially solved a memory problem.

Upgrading Java is a bit tricky so I am suspecting that you are still running the bundled Java.

Can you post your log file from `Tools / Open log directory` and attach the `messages.log`. It will contain your systems memory and which Java it is using.

The next problem is indeed with the Arc. I have tested sending this command to GRBL, FluidNC, GrblHAL and all of them are experiencing the same problem. Since the visualizer can render the model correctly (and also NCViewer) I am suspecting that this is a firmware problem. 
Can you share your FreeCAD project? I am curious if there are any settings that one could tweak in the post processor settings.

UGS has a feature to convert arcs to tiny line segments - "Arc Expander", when I activated this I can run the project:
[image]

gatornuke235, 2024-05-11:
Thanks for the clarification with the memory issue. This machine has 16GB of RAM, so that's probably not the issue. I've also attached the log file. I wound up uninstalling java and getting the latest 64bit offline install version from java.com and that seems to have fixed the freezing issue.

FreeCAD does indeed have a "split arcs" option in the properties for each operation path. I suspect this does the same as the "arc expander." I personally would prefer to leave the small proper arcs and just linearizing the ones with humongous radii, but that's personal preference I suppose.

Either way, I feel this issue is resolved and I'm closing the ticket. I appreciate your time. I just wanted to bring it to your attention in case you hadn't seen this yet.

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/15283796/messages.log)
[image]

closed 2024-05-11 by gatornuke235

breiler, 2024-05-12:
Thanks for the feedback.

Just a note, according to the log file UGS is still using the bundled Java at:
`C:\Users\eddyc\OneDrive\Desktop\ugsplatform-win\jdk\jdk-17.0.8.1+1-jre`

If you encounter more problems, try and move UGS to another folder outside OneDrive. The problems you are experienced could have been related to the program folder getting synced. 🤷🏻‍♂️

gatornuke235, 2024-05-13:
> If you encounter more problems, try and move UGS to another folder outside OneDrive. The problems you are experienced could have been related to the program folder getting synced. 🤷🏻‍♂️

Roger. Thanks.