winder/Universal-G-Code-Sender #2519: Unable to load Gcode, visualizer partially updates.

alum22, 2024-05-12:
### Version

2.1.6

### Hardware / Firmware

GRBL 1.1

### What happened

UGS works fine except when loading gcode with an .nc file. The visualizer updates to show the outer dimensions of the program, but no paths are show. The status changes and an error shows up in the lower right status bar. When I click it, it shows an error (see attached screenshot).

[image]

### How to reproduce

With UGS on, I navigate to File>Open. I open any .nc file (I generate my gcode programs from Fusion 360).

The Visualizer flashes and the error appears. The visualizer dimensions change to that of the program, but no paths are shown 0 lines are loaded.

### Operating System

Windows 10 Pro, Version 22H2

### Anything else

Everytime. I am unable to load an .NC file

breiler, 2024-05-13:
Can you share your log file. Open them menu "Tools > Open log folder" and attach the messages.log here.

Can you also share your gcode file.

breiler, 2024-05-13:
Looks like the settings have become corrupt, try deleting the folder `C:\Users\User\AppData\Roaming\ugsplatform` and start UGS again.

alum22, 2024-05-14:
Unfortunately, it is giving the same error.
[image]

breiler, 2024-05-14:
Can you post a new log file

alum22, 2024-05-14:
Yes, of course
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/15308497/messages.log)

breiler, 2024-05-14:
Still looks like the settings are corrupted.

UGS is looking for some files under C:\Users\User\Downloads\win64-ugs-platform-app-2.1.6\ which I assume is the old location. You have now moved all files to "C:\Program Files\ugsplatform-win\".

Shut down UGS and delete the folders: `C:\Users\User\AppData\Roaming\ugsplatform` and possible `C:\Users\User\AppData\Local\ugsplatform\Cache\var\cache` again.

alum22, 2024-05-14:
That did it!! Thank you so much!

closed 2024-05-14 by breiler